RetroPie/RetroPie-Setup #2268: lr-mame2010 and mame.ini file

msheehan79, 2018-01-23:
The current install script for lr-mame2010 puts the mame.ini file in /opt/retropie/libretrocores/lr-mame2010. 
However the core seems to be looking for this file in /home/pi. 

While most games seem to run fine without it, I ran into at least one issue that requires mame.ini to be located in the /home/pi folder. Mouse input is not recognized on some games (ex: Operation Thunderbolt) when mame.ini is not present. I made no changes to the default settings in the mame.ini file when testing.

As part of the install script should a copy of mame.ini get added to the /home/pi directory if the file is not already present, or are there some concerns or reasons for not doing this?

joolswills, 2018-01-23:
Sounds like that should be fixed upstream. Don't really want to put the ini file in the home directory.

Note - please use the forum first in future for things like this as detailed when opening an issue.

markwkidd, 2018-01-26:
> the core seems to be looking for this file in /home/pi

@msheehan79 that shouldn't be the case. Could you please file this bug in the libretro-mame2010 issue tracker if it isn't already listed?

msheehan79, 2018-01-26:
No worries. It was already raised 2 years ago regarding the cfg and nvram directories also getting created in the home directory. The mame.ini is expected to be in the same place. 

https://github.com/libretro/mame2010-libretro/issues/14

movisman, 2018-02-09:
Hi, actually yes I just noticed this too when installing the 2014 core for the first time.

@markwkidd if you come up with a solution, would you be kind enough to let me know? All the other MAME cores seem to put their files in a consistent location except this one. It would be great to get those folders shifted.

markwkidd, 2018-02-19:
@msheehan79 this turned out to be a bigger project than I expected. As always!

If you recompile mame2010 from source now, it will look for ini files in 'libretro saves folder/mame2010/ini'.

In fact, mame2010 now searches for everything from within the libretro structure. This is now briefly documented in the in the README at the bottom of the  ://github.com/libretro/mame2010-libretro

I have one more substantial PR pending which will autogenerate these folders: https://github.com/libretro/mame2010-libretro/pull/89

If you have the ability to compile your own cores from a PR branch, it would be great to make sure that these changes are working well on the rpi platform. :)

If not, give it a few days and the patch will hopefully be reviewed and merged into the master mame2010 branch.

msheehan79, 2018-02-19:
@markwkidd Indeed it seems so! I've been tracking your commits to the core this week. 

I was holding off updating until the auto folder gen code was added to try that as well, but if you want some feedback I can compile it from source later today and give it a whirl tonight.

Glad to see this core has gotten some love as of late to bring it more in line with the other Mame cores. To me it seems to have gotten a bad rap, as I've found a number of different games that either didn't run or were more accurate on mame2010 than on 2003 and still run at 60fps.

markwkidd, 2018-02-19:
If you do get time to test my branch, I would be appreciated.

I'm not sure I have anyone else lined up to test, and as you may have seen with my earlier attempt at making directories, just because it's working on my system doesn't mean it won't break the buildbot!

It can't hut my case when asking Twinaphex to merge the PR if I have testing from more than one platform.

Speaking of mame2010's bad rap: the next thing that's caught my eye is adding support for more than two players. I've started that process, but then I spent more time on fixing these paths than I intended so I will have to take a break from that. I'm hoping in a few days to try to get r-type to help with it. I think he could probably do it in an hour :)

msheehan79, 2018-02-19:
@markwkidd Sure thing, I'll give your branch a shot later today and let you know!

msheehan79, 2018-02-19:
Hi @markwkidd - Built the core today and I can confirm the config directories do now build in the same folders as mame2003 does, so the ~RetroPie/roms/arcade (in my case) folder now has a new folder named mame2010 with the various config directories. 

Also confirmed the samples, artwork, crosshairs and fonts directories were created in the bios directory, again same as mame2003.

I only tested 2 games so far but both worked just fine and created cfg and nvram files for them. Next step is to port across the existing configs I had previously in the home directory. 

I also still need to test the original reason for the problem report, which was Operation Thunderbolt did not recognize mouse input without mame.ini being in the home directory.

msheehan79, 2018-02-19:
Have now moved my cfg and nvram and other relevant folders from the "old" locations in the home directory to their new homes and all seems to work well with migrating config data. 

As for the original issue report, once I moved the mame.ini file into the ~/RetroPie/roms/arcade/mame2010/ini/ directory, Operation Thunderbolt recognized mouse input again. 

I would suggest once all the PRs get merged it would be advisable to also see if we can update the RetroPie Setup script to put the mame.ini file into this new folder if at all possible, so users don't have to manually move it there after installing the core. Not sure if it presents a problem in that depending upon user preference Mame2010 roms could be in either the "arcade" folder or the "mame-libretro" folder.

markwkidd, 2018-02-20:
Thanks for your work testing this @msheehan79 ! I agree it would be great to help users migrate to the new paths.

If nothing else, maybe we could come up with a few lines of bash script that people could copy paste themselves.

markwkidd, 2018-03-06:
@msheehan79 could this issue be closed now?

msheehan79, 2018-03-08:
@markwkidd I haven't had much time to retest as I just started a new job this week, but I think with the recent updates you've done to the core to update the paths and embed the mame.ini file it is probably OK to close this. 

I'd feel better retesting just to err on the side of caution but it's likely going to be at least another week or two until I get my Pi back together to recompile. I'll close for now and if I do run into any issues I can reopen then. Thanks!

closed 2018-03-08 by msheehan79