RetroPie/RetroPie-Setup #2269: Illegal usage of your software by a french company, NESPi ?

JackoboLeChocobo, 2018-01-25:
According to that website, this French company sells Raspberry PI cases with Retropie software included with emulators from you and libretro ( and SD cards that have games included too ?! )

So this company violates the licences of RetroPie and Libretro by selling their products. My twin brother has warned me of this. Having an account on Github we wish to warn you.

Here is informations about the French company if you want to contact them about that violation of the licences :  

-> Official page of the product : https://boutique-retrogaming.com/

-> Products : https://boutique-retrogaming.com/categorie-produit/console-de-jeu-retro-nespi-case/

-> Products that violate the licence for instance : 
=> https://boutique-retrogaming.com/boutique/jeux-retro/nintendo/console-nespi-16go-innext/
=> https://boutique-retrogaming.com/boutique/jeux-retro/nintendo/console-nespi-16go/
=> https://boutique-retrogaming.com/boutique/jeux-retro/nintendo/nespi-case-32go/
=> https://boutique-retrogaming.com/boutique/jeux-retro/nintendo/console-nespi-128go/
=> https://boutique-retrogaming.com/boutique/jeux-retro/nintendo/console-nespi-32go-2-manettes-xbox/
=> https://boutique-retrogaming.com/boutique/jeux-retro/nintendo/console-nespi-32go-4-manettes/
=> https://boutique-retrogaming.com/boutique/jeux-retro/nintendo/console-nespi-128go-xbox/
=> https://boutique-retrogaming.com/boutique/jeux-retro/nintendo/console-nespi-128go-4-manettes/

-> Facebook Page : https://www.facebook.com/Boutique-Retrogaming-1860673187510896/

-> Company  : 
Name : Boutique Retrogaming
Address : 15 rue des pêcheurs, 59240 Dunkerque - France
Phone number ( France ) : 06 34 67 76 04
Email address : contact@boutique-retrogaming.com

You can download on the ticket the PDF documents from the company that we found about their products : 
[NESPi_Liste_Jeux_128GB.pdf](https://github.com/RetroPie/RetroPie-Setup/files/1665254/NESPi_Liste_Jeux_128GB.pdf)
[NESPi_Notice_Utilisateur.pdf](https://github.com/RetroPie/RetroPie-Setup/files/1665255/NESPi_Notice_Utilisateur.pdf)
[NESPi_Liste_Jeux_32GB.pdf](https://github.com/RetroPie/RetroPie-Setup/files/1665256/NESPi_Liste_Jeux_32GB.pdf)
[NESPi_Liste_Jeux_16GB.pdf](https://github.com/RetroPie/RetroPie-Setup/files/1665257/NESPi_Liste_Jeux_16GB.pdf)

Note : This ticket will be written for RetroArch and RetroPie, both are concerned by this issue.

JackoboLeChocobo, 2018-01-26:
Updated informations about the company if needed

joolswills, 2018-01-26:
Thanks for the info. Will close, as prefer not to use the issue tracker for things like this.

closed 2018-01-26 by joolswills

JackoboLeChocobo, 2018-01-31:
Even with proofs, nothing in France want help to stop the sell of these products. Solution in France : paid between 1000 and 2000 $ a lawyer. I suggest you to see solutions in your side about the illegal use of your software.