RetroPie/RetroPie-Setup #2277: Could not successfully build mupen64plus, Ubuntu 16.04

doedeler, 2018-01-29:
Hi, 

my OS is Ubuntu 16.04 32-bit:
```
retro@retro:~$ uname -a
Linux retro 4.13.0-32-generic #35~16.04.1-Ubuntu SMP Thu Jan 25 10:13:25 UTC 2018 i686 i686 i686 GNU/Linux
```

And i followed the installation guide for RetroPie on Ubuntu. 
SNES, NES, SEGA Emulator work perfectly, but i get an error message after Basic install that says: 

> Could not successfully build mupen64plus - N64 emulator MUPEN64Plus
> (GliedeN64/projects/cmake/plugin/release/mupen64plus-video-GLideN64.so not found)

Here is the end of mupen64plus installation log:
```
/usr/bin/ld: i386:x86-64 architecture of input file `/home/retro/RetroPie-Setup/tmp/build/mupen64plus/GLideN64/src/GLideNHQ/lib/libz.a(inffast.o)' is incompatible with i386 output
collect2: error: ld returned 1 exit status
CMakeFiles/mupen64plus-video-GLideN64.dir/build.make:2572: die Regel für Ziel „plugin/release/mupen64plus-video-GLideN64.so“ scheiterte
make[2]: *** [plugin/release/mupen64plus-video-GLideN64.so] Fehler 1
CMakeFiles/Makefile2:68: die Regel für Ziel „CMakeFiles/mupen64plus-video-GLideN64.dir/all“ scheiterte
make[1]: *** [CMakeFiles/mupen64plus-video-GLideN64.dir/all] Fehler 2
Makefile:83: die Regel für Ziel „all“ scheiterte
make: *** [all] Fehler 2
~/RetroPie-Setup/tmp/build/mupen64plus ~/RetroPie-Setup
Removing additional swap
~/RetroPie-Setup
Could not successfully build mupen64plus - N64 emulator MUPEN64Plus (GLideN64/projects/cmake/plugin/release/mupen64plus-video-GLideN64.so not found).
~/RetroPie-Setup/tmp/build/lr-atari800 ~/RetroPie-Setup

```
I am sorry that it is in german, i hope you can still see whats going wrong. 

I installed RetroPie 3 times new, i updated the script and i updated the packages.. nothing works. 

I hope you guys can help me. Have a nice day

MJB-G, 2018-05-06:
Hi,
Same problem. My OS : Ubuntu 18.04 for Odroid XU4.
Thank for your help.

gizmo98, 2019-08-08:
Should be fixed with this commit:
https://github.com/RetroPie/RetroPie-Setup/commit/111043b65924504dab415a32ebe38877e0f1ed8f#diff-fe35047c4f73377c42592c2271157752

Anon0ne, 2020-04-25:
I'm still having this issue but on Lubuntu/Atomic Pi hardware. I am updated on everything and it worked for two days until a reboot occured now it won't load the emulator. Went to reinstall it and this shows up

MsMythchief, 2020-09-20:
Getting this error installing on Linux4Tegra Ubuntu 18.04

`Log started at: Sun Sep 20 15:17:28 MDT 2020

RetroPie-Setup version: 4.6.7 (b6bb01b)
System: jetson-nano (aarch64) - Ubuntu 18.04.5 LTS - Linux nano-desktop 4.9.140-tegra #1 SMP PREEMPT Thu Jun 25 21:25:44 PDT 2020 aarch64 aarch64 aarch64 GNU/Linux

= = = = = = = = = = = = = = = = = = = = =
Installing dependencies for 'retroarch' : RetroArch - frontend to the libretro emulator cores - required by all lr-* emulators
= = = = = = = = = = = = = = = = = = = = =

~/RetroPie-Setup/tmp/build/retroarch ~/RetroPie-Setup

= = = = = = = = = = = = = = = = = = = = =
Getting sources for 'retroarch' : RetroArch - frontend to the libretro emulator cores - required by all lr-* emulators
= = = = = = = = = = = = = = = = = = = = =

git clone --recursive --depth 1 --branch v1.8.8 "https://github.com/libretro/RetroArch.git" "/home/nano/RetroPie-Setup/tmp/build/retroarch"
Cloning into '/home/nano/RetroPie-Setup/tmp/build/retroarch'...
Note: checking out '9552f87157120252a9fad2e2c108ab08c2d3735e'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

Checking out files:  62% (5330/8547)   
Checking out files:  63% (5385/8547)   
Checking out files:  64% (5471/8547)   
Checking out files:  65% (5556/8547)   
Checking out files:  66% (5642/8547)   
Checking out files:  67% (5727/8547)   
Checking out files:  68% (5812/8547)   
Checking out files:  69% (5898/8547)   
Checking out files:  70% (5983/8547)   
Checking out files:  71% (6069/8547)   
Checking out files:  72% (6154/8547)   
Checking out files:  73% (6240/8547)   
Checking out files:  74% (6325/8547)   
Checking out files:  75% (6411/8547)   
Checking out files:  76% (6496/8547)   
Checking out files:  77% (6582/8547)   
Checking out files:  78% (6667/8547)   
Checking out files:  79% (6753/8547)   
Checking out files:  80% (6838/8547)   
Checking out files:  81% (6924/8547)   
Checking out files:  82% (7009/8547)   
Checking out files:  83% (7095/8547)   
Checking out files:  84% (7180/8547)   
Checking out files:  85% (7265/8547)   
Checking out files:  86% (7351/8547)   
Checking out files:  87% (7436/8547)   
Checking out files:  88% (7522/8547)   
Checking out files:  89% (7607/8547)   
Checking out files:  90% (7693/8547)   
Checking out files:  90% (7695/8547)   
Checking out files:  91% (7778/8547)   
Checking out files:  92% (7864/8547)   
Checking out files:  93% (7949/8547)   
Checking out files:  94% (8035/8547)   
Checking out files:  95% (8120/8547)   
Checking out files:  96% (8206/8547)   
Checking out files:  97% (8291/8547)   
Checking out files:  98% (8377/8547)   
Checking out files:  99% (8462/8547)   
Checking out files: 100% (8547/8547)   
[... 2533 lines cut ...]
make: Leaving directory '/home/nano/RetroPie-Setup/tmp/build/mupen64plus/mupen64plus-video-glide64mk2/projects/unix'
current revision "b6bb01b"
last build revision "1f5f987"
~/RetroPie-Setup/tmp/build/mupen64plus/GLideN64/projects/cmake ~/RetroPie-Setup/tmp/build/mupen64plus ~/RetroPie-Setup
Interprocedural optimizations enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nano/RetroPie-Setup/tmp/build/mupen64plus/GLideN64/projects/cmake
[  2%] Built target osal
[ 14%] Built target GLideNHQ
[ 15%] Generate Git version
current revision "1f5f987"
last build revision "b6bb01b"
Scanning dependencies of target mupen64plus-video-GLideN64
[ 15%] Building CXX object CMakeFiles/mupen64plus-video-GLideN64.dir/Debugger.cpp.o
[ 15%] Building CXX object CMakeFiles/mupen64plus-video-GLideN64.dir/Graphics/OpenGLContext/mupen64plus/mupen64plus_DisplayWindow.cpp.o
[ 16%] Linking CXX shared library plugin/Release/mupen64plus-video-GLideN64.so
[100%] Built target mupen64plus-video-GLideN64
~/RetroPie-Setup/tmp/build/mupen64plus ~/RetroPie-Setup
Removing additional swap
~/RetroPie-Setup
Could not successfully build mupen64plus - N64 emulator MUPEN64Plus (mupen64plus-rsp-z64/projects/unix/mupen64plus-rsp-z64.so not found).

Log ended at: Sun Sep 20 15:34:03 MDT 2020
Total running time: 0 hours, 16 mins, 35 secs`

MsMythchief, 2020-09-20:
[rps_2020-09-20_151728.log](https://github.com/RetroPie/RetroPie-Setup/files/5252791/rps_2020-09-20_151728.log)
Also getting this error on my Jetson Nano.

joolswills, 2020-09-21:
Use the forum first

joolswills, 2021-06-27:
Closing as Ubuntu 16.04 isn't supported anymore and it should build fine on later Ubuntu versions.

closed 2021-06-27 by joolswills