RetroPie/RetroPie-Setup #2285: Correct 2 player Twin Stick games with Xin-Mo encoder

2Grateful, 2018-02-08:
Would be great to get the Xin-Mo encoder working 'properly' .  

Works with Windows/Emulation Station/Retroarch just fine for second stick (Player2 Stick) to be mapped for games like Robotron and Smash TV

The same is not the case in Retropie.  Does not recognize Joypad2 for second control.  Would love to see the Xin-Mo finally fixed to show up and work properly.  

Thanks

joolswills, 2018-02-08:
Did you open a thread on the forum first as detailed when opening an issue ?

2Grateful, 2018-02-08:
New to reporting issues with this.  So no.

joolswills, 2018-02-09:
No worries. But please start a help topic on the RetroPie forum first as it may be an issue addressable with a config change. Eg a kernel parameter.

closed 2018-02-09 by joolswills

2Grateful, 2018-02-09:
Will do. Thank you