RetroPie/RetroPie-Setup #2289: Forum sign-up mails not being sent out

hholst80, 2018-02-10:
forum sign-up mails are not being sent out as they should. Checked spam and inbox and nothing there.

joolswills, 2018-02-10:
Can't see any problem. Your signup mail was delivered to the google mail server.

```
2018-02-10 00:18:30 1ekIs9-0001Fg-6L => YOUREMAIL@gmail.com R=dnslookup T=remote_smtp H=gmail-smtp-in.l.go
ogle.com [2a00:1450:400c:c04::1b] X=TLS1.2:ECDHE_RSA_CHACHA20_POLY1305:256 CV=yes DN="C=US,ST=California,
L=Mountain View,O=Google Inc,CN=mx.google.com" C="250 2.0.0 OK 1518221910 p70si145485wma.88 - gsmtp"
```

joolswills, 2018-02-10:
Must be in your mailbox somewhere unless gmail lost it (but that's unlikely). You can request another one from the user profile page.

closed 2018-02-10 by joolswills

hholst80, 2018-02-10:
You are right it was sorted by Gmail under one of the other inboxes. I was too tired to think of that possibility yesterday. Sorry about that. Thank you for looking into it.