RetroPie/RetroPie-Setup #2295: lr-mgba isn't just a Game Boy Advance (gba) emulator, it also emulates Game Boy (gb) and Game Boy Color (gbc).

vaguerant, 2018-02-14:
I've added it to the `/opt/retropie/configs/gb/emulators.cfg` and `/opt/retropie/configs/gbc/emulators.cfg` files on my end. I tried to find where these are on the repo but had no luck or else I would have made a pull request, sorry.

While native mGBA supports the Game Boy, Super Game Boy, Game Boy Color and Game Boy Advance, the libretro version is admittedly more GBA-focussed. It doesn't seem to have support for the Super Game Boy, or BIOSes for the other Game Boys besides Advance. Still, it's a very solid option for Game Boy/Color games and already included in RetroPie, so why not include the option to run GB/C games with it?

Darknior, 2018-02-14:
Yes you are real ... it can be cool to add it for the two other gameBoy.

https://github.com/libretro/mgba

joolswills, 2018-02-14:
What advantages does it have over the other emulators ? Have you tested it?

Darknior, 2018-02-15:
Like you write it, i have add it to my RETROPIE easily :)
I don't try playing games with it for the moment, but an other emu is always cool to prevent a bug of one emulator on a game and trying with an other one.
And like you write mGBA have an active development :)
For me it's a good idea to add it to GB and GBC.

vaguerant, 2018-04-07:
Just wanted to note here that upstream at libretro/mgba, mGBA has been updated to be in-line with current standalone mGBA, meaning it now supports (Super) Game Boy (Color) BIOSes, full SGB border and palette support, Game Boy Color enhanced palettes for original Game Boy games, etc. I'd say this will make mGBA easily the best Game Boy/Color/Advance emulator available in RetroPie in terms of its capabilities. These updates are not in a stable release yet; RetroPie users can update their lr-mgba from source to get these updates right now.

Speed-wide, Game Boy emulation in mGBA seems to run about 75% of the speed of Gambatte from a quick test. On a Pi 3, this is the difference between 430 FPS and 587 FPS for Game Boy Color emulation. i.e. The difference is unlikely to be noticed at all on Pi 3, but I'm not sure how low-spec a setup RetroPie's defaults are intended to target. I suspect mGBA's Game Boy emulation is probably fast enough for any platform likely to be used to run RetroPie, but I don't own any e.g. Pi Zeros to confirm this. If mGBA were to be made the new default for Game Boy/Color, I think the change people would be most likely to notice is Super Game Boy support out of the box for RetroPie, which Gambatte doesn't provide.

Even if it's not made the new default, these updates should certainly secure mGBA at least a place in the core selection for those platforms, which it doesn't currently have.

hhromic, 2018-04-12:
now that `lr-mgba` is updated to support GB/GBC/GBA etc. fully, maybe the best thing to do is to at least allow `lr-mgba` to be an available emulator for those consoles (e.g add it to `/opt/retropie/configs/gb/emulators.cfg` as suggested in the issue), but without making it the default.
This way people who wants to use mgba for all these consoles in fast hardware, e.g. rpi3, has the option out-of-the-box without harm to the other users who don't.

closed 2018-04-20 by joolswills

hyperM09, 2019-04-25:
I know this threat is a little bit old, but i can't connect the gb roms with lr-mgba.
I thought i only have to add a line in the gb's emulators.cfg

**so thats mine without any changes**
lr-gambatte = "/opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-gambatte/gambatte_libretro.so --config /opt/retropie/configs/gb/retroarch.cfg %ROM%"
default = "lr-gambatte"
lr-tgbdual = "/opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-tgbdual/tgbdual_libretro.so --config /opt/retropie/configs/gb/retroarch.cfg %ROM%"

**so i thought i can easily add a new line to connect it**
lr-mgba =
"/opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-mgba/mgba_libretro.so --
config /opt/retropie/configs/gb/retroarch.cfg %ROM%"

the emulator appears in the rom-settings, but the game doesnt start.
I'm afraid that I missed something somewhere
maybe you can help me.

commit a58f172f (2018-04-20): lr-mgba.sh: enable lr-mgba as alternative also for Game Boy and Game Boy Color (fixes #2295)