RetroPie/RetroPie-Setup #2296: Wolf4SDL Not Launching 'Spear of Destiny' (Mission 1)

mediamoshpit, 2018-02-15:
This is in reference to a potential issue [first mentioned here](https://retropie.org.uk/forum/topic/16260/wolf4sdl-launching-spear-of-destiny-mission-1). 

I believe the current mission one launch command of

`/opt/retropie/ports/wolf4sdl/bin/wolf4sdl-spear --mission 1`

might need to be

`/opt/retropie/ports/wolf4sdl/bin/wolf4sdl-spear`

psyke83, 2018-02-15:
I can't reproduce. Please post the output of /dev/shm/runcommand.log after trying to launch mission 1.

Posting the contents of your wolf4sdl folder may also help troubleshoot. Mission 1, 2 and 3 work properly for me after copying the ```*.SOD``` and ```*.SD*``` files from my Steam installation and renaming to lowercase.

```
pi@retropie:~/RetroPie/roms/ports/wolf3d $ ls -l
total 10260
-rw-r--r-- 1 pi pi    1072 Jan 28 02:29 audiohed.sod
-rw-rw-r-- 1 pi pi    1156 Dec  3  1992 audiohed.wl1
-rw-r--r-- 1 pi pi  328620 Jan 28 02:29 audiot.sod
-rw-rw-r-- 1 pi pi  132613 Dec  3  1992 audiot.wl1
-rw-rw-r-- 1 pi pi   34101 Dec 22  1992 catalog.exe
-rw-r--r-- 1 pi pi     522 Jan 28 02:51 config.sod
-rw-r--r-- 1 pi pi   68290 Jan 28 02:29 gamemaps.sd1
-rw-r--r-- 1 pi pi  135024 Jan 28 02:29 gamemaps.sd2
-rw-r--r-- 1 pi pi  148534 Jan 28 02:29 gamemaps.sd3
-rw-r--r-- 1 pi pi  148534 Jan 28 02:50 gamemaps.sod
-rw-rw-r-- 1 pi pi   27425 Dec  3  1992 gamemaps.wl1
-rw-r--r-- 1 pi pi     402 Jan 28 02:29 maphead.sd1
-rw-r--r-- 1 pi pi      86 Jan 28 02:29 maphead.sd2
-rw-r--r-- 1 pi pi      86 Jan 28 02:29 maphead.sd3
-rw-r--r-- 1 pi pi      86 Jan 28 02:50 maphead.sod
-rw-rw-r-- 1 pi pi     402 Dec  3  1992 maphead.wl1
-rw-rw-r-- 1 pi pi    2564 May 22  1992 order.frm
-rw-rw-r-- 1 pi pi    6090 Dec 17  1992 read1st.txt
-rw-rw-r-- 1 pi pi    1221 Dec 22  1992 vendor.doc
-rw-r--r-- 1 pi pi    1024 Jan 28 02:29 vgadict.sod
-rw-rw-r-- 1 pi pi    1024 Dec  3  1992 vgadict.wl1
-rw-r--r-- 1 pi pi  947979 Jan 28 02:29 vgagraph.sod
-rw-rw-r-- 1 pi pi  326568 Dec  3  1992 vgagraph.wl1
-rw-r--r-- 1 pi pi     510 Jan 28 02:29 vgahead.sod
-rw-rw-r-- 1 pi pi     471 Dec  3  1992 vgahead.wl1
-rw-r--r-- 1 pi pi 1616544 Jan 28 02:29 vswap.sd1
-rw-r--r-- 1 pi pi 1879712 Jan 28 02:29 vswap.sd2
-rw-r--r-- 1 pi pi 1881760 Jan 28 02:29 vswap.sd3
-rw-r--r-- 1 pi pi 1881760 Jan 28 02:50 vswap.sod
-rw-rw-r-- 1 pi pi  742912 Dec  3  1992 vswap.wl1
-rw-rw-r-- 1 pi pi  109959 Dec  3  1992 wolf3d.exe
```

mediamoshpit, 2018-02-15:
In runcommand.log, I get "NO SPEAR OF DESTINY DATA FILES TO BE FOUND!", which is strange, because without the addition of --mission 1, the game loads into the first mission of SoD just fine.

The contents of my wolf4sdl are:
```
pi@retropie:~ $ ls -l /home/pi/RetroPie/roms/ports/wolf3d/
total 12576
-rwxr-xr-x 1 pi pi   17154 Jul 22  2017 10.cmf
-rwxr-xr-x 1 pi pi   23427 Jul 22  2017 11.cmf
-rwxr-xr-x 1 pi pi   16760 Jul 22  2017 12.cmf
-rwxr-xr-x 1 pi pi    6527 Jul 22  2017 13.cmf
-rwxr-xr-x 1 pi pi   18748 Jul 22  2017 14.cmf
-rwxr-xr-x 1 pi pi   19489 Jul 22  2017 15.cmf
-rwxr-xr-x 1 pi pi   21054 Jul 22  2017 16.cmf
-rwxr-xr-x 1 pi pi    9792 Jul 22  2017 1.cmf
-rwxr-xr-x 1 pi pi   21954 Jul 22  2017 2.cmf
-rwxr-xr-x 1 pi pi    6210 Jul 22  2017 3.cmf
-rwxr-xr-x 1 pi pi   14394 Jul 22  2017 4.cmf
-rwxr-xr-x 1 pi pi   12398 Jul 22  2017 5.cmf
-rwxr-xr-x 1 pi pi   13988 Jul 22  2017 6.cmf
-rwxr-xr-x 1 pi pi   22740 Jul 22  2017 7.cmf
-rwxr-xr-x 1 pi pi    2406 Jul 22  2017 8.cmf
-rwxr-xr-x 1 pi pi   17667 Jul 22  2017 9.cmf
-rwxr-xr-x 1 pi pi   50605 Jul 22  2017 advanced.pcx
-rwxr-xr-x 1 pi pi    1072 Jul 22  2017 audiohed.sod
-rwxr-xr-x 1 pi pi    1156 Jul 22  2017 audiohed.wl1
-rwxr-xr-x 1 pi pi    1156 Jul 22  2017 audiohed.wl6
-rwxr-xr-x 1 pi pi  328620 Jul 22  2017 audiot.sod
-rwxr-xr-x 1 pi pi  132613 Jul 22  2017 audiot.wl1
-rwxr-xr-x 1 pi pi  320209 Jul 22  2017 audiot.wl6
-rwxr-xr-x 1 pi pi   14635 Jul 22  2017 blood.voc
-rwxr-xr-x 1 pi pi   34101 Jul 22  2017 catalog.exe
-rwxr-xr-x 1 pi pi     732 Jul 22  2017 config.sod
-rwxr-xr-x 1 pi pi   54678 Jul 22  2017 create.pcx
-rwxr-xr-x 1 pi pi   26494 Jul 22  2017 credits.pcx
-rwxr-xr-x 1 pi pi   44674 Jul 22  2017 devil.voc
-rwxr-xr-x 1 pi pi    4604 Jul 22  2017 dog.voc
-rwxr-xr-x 1 pi pi  149605 Jul 22  2017 gamegen.exe
-rwxr-xr-x 1 pi pi  135024 Jul 22  2017 gamemaps.sd2
-rwxr-xr-x 1 pi pi  148534 Jul 22  2017 gamemaps.sd3
-rwxr-xr-x 1 pi pi   68290 Jul 22  2017 gamemaps.sod
-rwxr-xr-x 1 pi pi   27425 Jul 22  2017 gamemaps.wl1
-rwxr-xr-x 1 pi pi  150652 Jul 22  2017 gamemaps.wl6
-rwxr-xr-x 1 pi pi   37302 Jul 22  2017 intro.pcx
-rwxr-xr-x 1 pi pi    5527 Jul 22  2017 machine.voc
-rwxr-xr-x 1 pi pi   37928 Jul 22  2017 main.pcx
-rwxr-xr-x 1 pi pi      86 Jul 22  2017 maphead.sd2
-rwxr-xr-x 1 pi pi      86 Jul 22  2017 maphead.sd3
-rwxr-xr-x 1 pi pi     402 Jul 22  2017 maphead.sod
-rwxr-xr-x 1 pi pi     402 Jul 22  2017 maphead.wl1
-rwxr-xr-x 1 pi pi     402 Jul 22  2017 maphead.wl6
-rwxr-xr-x 1 pi pi    2564 Jul 22  2017 order.frm
-rwxr-xr-x 1 pi pi    6090 Jul 22  2017 read1st.txt
-rwxr-xr-x 1 pi pi    5663 Jul 22  2017 scream.voc
-rwxr-xr-x 1 pi pi    8315 Jul 22  2017 sod.exe
-rwxr-xr-x 1 pi pi    1221 Jul 22  2017 vendor.doc
-rwxr-xr-x 1 pi pi    1024 Jul 22  2017 vgadict.sod
-rwxr-xr-x 1 pi pi    1024 Jul 22  2017 vgadict.wl1
-rwxr-xr-x 1 pi pi    1024 Jul 22  2017 vgadict.wl6
-rwxr-xr-x 1 pi pi  947979 Jul 22  2017 vgagraph.sod
-rwxr-xr-x 1 pi pi  326568 Jul 22  2017 vgagraph.wl1
-rwxr-xr-x 1 pi pi  334506 Jul 22  2017 vgagraph.wl6
-rwxr-xr-x 1 pi pi     510 Jul 22  2017 vgahead.sod
-rwxr-xr-x 1 pi pi     471 Jul 22  2017 vgahead.wl1
-rwxr-xr-x 1 pi pi     486 Jul 22  2017 vgahead.wl6
-rwxr-xr-x 1 pi pi 1879712 Jul 22  2017 vswap.sd2
-rwxr-xr-x 1 pi pi 1881760 Jul 22  2017 vswap.sd3
-rwxr-xr-x 1 pi pi 1616544 Jul 22  2017 vswap.sod
-rwxr-xr-x 1 pi pi  742912 Jul 22  2017 vswap.wl1
-rwxr-xr-x 1 pi pi 1545400 Jul 22  2017 vswap.wl6
-rwxr-xr-x 1 pi pi  109959 Jul 22  2017 wolf3d.exe
```

psyke83, 2018-02-15:
Your installation doesn't use the .sd1 extension to mark files as mission 1. The question is whether your source is incorrect, or if Steam's packaging does something non-standard with the original files.

For the moment, can you try renaming your .sod files to .sd1 and let me know if the ```--mission 1``` launcher works?

mediamoshpit, 2018-02-15:
After renaming, I got "Can't open /home/pi/RetroPie/roms/ports/wolf3d/vgadict.sod!". I renamed it back from vgadict.sd1 and Wolf4SDL progressively threw an error asking for the .sod counterparts of all the renamed .sd1 files.

psyke83, 2018-02-20:
IMHO, I don't think that any changes to the scriptmodule are necessary, but the documentation may need updating.

Here's the complete list of files that need to be copied from your Spear of Destiny source:
```
audiohed.sod
audiot.sod
config.sod
gamemaps.sd1
gamemaps.sd2
gamemaps.sd3
maphead.sd1
maphead.sd2
maphead.sd3
vgadict.sod
vgagraph.sod
vgahead.sod
vswap.sd1
vswap.sd2
vswap.sd3
```
Apologies for the prior advice, but of the existing .sod files, you only needed to rename these three files:
```
gamemaps.sod -> gamemaps.sd1
maphead.sod -> maphead.sd1
vswap.sod -> vswap.sd1
```

If you can verify the above information is accurate, I'll update the wiki page to clarify the situation.

mediamoshpit, 2018-02-20:
Bingo, bango, bongo! That did the trick. Everything seems to be loading fine now. Although, I do have to ask, do missions two and three load within a surrounding black frame on your install?

psyke83, 2018-02-26:
I've updated the wiki, so this issue can be closed. Thanks.

psyke83, 2018-02-27:
Apologies, I overlooked your question. It seems that mission 2 and 3 don't launch with the required "SDL1_VIDEODRIVER=dispmanx" environment variable, so they're not being configured to use dispmanx correctly. Will reopen this issue to remind me to sort out a fix.

mediamoshpit, 2018-02-27:
Not a problem. Thanks for looking into it.

psyke83, 2018-03-08:
@mediamoshpit. It should be sorted now (via #2319), but you'll need to update your RetroPie-Setup script and re-install. I've improved the script so that it generates separate launchers - no need to switch between episodes via runcommand, etc. Will close this for the final time (hopefully!).

Until the documentation is updated, just keep in mind that the port launchers are generated only when the appropriate files are detected. It detects which launcher to create based on the existence of the ```vswap.{wl1,wl6,sd1,sd2,sd3}``` files, and the launcher will handle which binary needs to be launched for each one based on the checksum. As long as the md5sums match what's currently on the wiki, you should be OK.

closed 2018-03-08 by psyke83

mediamoshpit, 2018-03-08:
Everything went off without a hitch and all the variation look and play beautifully.