RetroPie/RetroPie-Setup #2305: Installer script installs as root then won't boot

wgx, 2018-02-21:
Hello! I did the installer script on a fresh Raspbian install, it installs as root - then when booted emulationstation can't start as root. I just used defaults for everything.

wgx, 2018-02-21:
I followed these instructions: https://github.com/RetroPie/RetroPie-Setup/blob/master/README.md

After you do dist-upgrade you need to reboot. Also I made the setup script executable with chmod +x before I ran it. It worked the second time (from another fresh install of Raspbian Lite).

joolswills, 2018-02-21:
These instructions are more detailed - https://retropie.org.uk/docs/Manual-Installation/

the set-up script should be executable by default. Permissions looks correct here. Please use the forum first for advice as detailed when opening a ticket.

joolswills, 2018-02-21:
(Note you should not install as root, but via sudo under the pi user)

closed 2018-02-21 by joolswills

wgx, 2018-02-21:
> These instructions are more detailed

Yes they are. My point is the readme should reflect the same instructions as people following the readme in isolation won't know the stuff in the other document.

joolswills, 2018-02-21:
the basic readme instructions work here - and they do link to the main website for further information.