RetroPie/RetroPie-Setup #2306: Xpad Driver couldt Install

BoBBer446, 2018-02-21:
Hey, at thenlastest Version i cant Install xpad Driver, Herr isnlog File
```
Log started at: Wed Feb 21 16:48:37 CET 2018                                                                 
                                                                                                             
RetroPie-Setup version: 4.3.11 (abd0bd74)                                                                    
System: Linux osmc 4.9.29-15-osmc #1 SMP PREEMPT Sat Jan 20 21:27:28 UTC 2018 armv7l GNU/Linux               
                                                                                                             
= = = = = = = = = = = = = = = = = = = = =                                                                    
Installing dependencies for 'xpad' : Updated Xpad Linux Kernel driver                                        
= = = = = = = = = = = = = = = = = = = = =                                                                    
                                                                                                             
Did not find needed package(s): dkms raspberrypi-kernel-headers. I am trying to install them now.            
Get:1 http://security.debian.org stretch/updates InRelease [63.0 kB]                                         
Ign:2 http://ftp.debian.org/debian stretch InRelease                                                         
Get:3 http://ftp.debian.org/debian stretch-updates InRelease [91.0 kB]                                       
Hit:5 http://archive.itimmer.nl/raspbian/moonlight stretch InRelease                                         
Hit:6 http://ftp.debian.org/debian stretch Release                                                           
Hit:4 http://ftp.fau.de/osmc/osmc/apt stretch InRelease                                                      
Fetched 154 kB in 2s (55.5 kB/s)                                                                             
Reading package lists...                                                                                     
Reading package lists...                                                                                     
Building dependency tree...                                                                                  
Reading state information...                                                                                 
E: Unable to locate package raspberrypi-kernel-headers                                                       
Could not install package(s): dkms raspberrypi-kernel-headers.                                               
                                                                                                             
Log ended at: Wed Feb 21 16:48:47 CET 2018                                                                   
Total running time: 0 hours, 0 mins, 10 secs                                                                 
osmc@osmc:~/RetroPie-Setup/logs$
```

joolswills, 2018-02-21:
Sorry - this is not designed to work with OSMC.

BoBBer446, 2018-02-21:
All is working fine, xboxdrv is installing fine but dont works.

Maybe you can Help me to Install xpad
Raspberry Pi kernel headers package cannot be installed

joolswills, 2018-02-21:
You can't. OSMC package their own kernel and headers. It may work with changes but right now, this will not work with osmc.

closed 2018-02-21 by joolswills