RetroPie/RetroPie-Setup #2320: updating RetroPie re-installs core packages previously removed

hhromic, 2018-02-27:
When updating RetroPie using the normal procedure with RetroPie-Setup script, **core** packages such as `bashwelcometweak` or `retropiemenu` that were previously uninstalled (also using the RetroPie-Setup script) are re-installed again.

In other words, if I remove e.g. `retropiemenu` and then update my installation using the RetroPie-Setup script, the `retropiemenu` package gets re-installed and appears again in ES, forcing me to remove it again.
This does not happen with emulators or driver packages, where the upgrade respects their installation state, i.e. "uninstalled".

Perhaps RetroPie-Setup should also respect the state for core packages and leave them uninstalled if the user removed them manually?
Thanks for the great work in RetroPie.

joolswills, 2018-02-27:
Please start a topic on the forum first as detailed when opening an issue. I cannot reproduce the issue also.

hhromic, 2018-02-27:
Oh sorry for that. I'll report in the forum with steps to reproduce.
Thanks!

joolswills, 2018-02-27:
I was doing some testing and managed to reproduce the issue - I will sort it. Still prefer things to start on the forum as often issues are related to configuration etc (for example because someone has a custom es_systems.cfg)

joolswills, 2018-02-27:
The issue was specific to the retropiemenu due to a fix for upgrading old installs to the new packaging method, but I am not sure of any problems with other modules like the bashwelcometweak.

closed 2018-02-27 by joolswills

hhromic, 2018-02-27:
great you sorted it out for `retropiemenu` !
I do experience the same effect with `bashwelcometweak` in my clean setup.

I checked your fix and I think the same problem applies for `bashwelcometweak`: there is no `$md_inst` created for this module because it just simply appends and removes a bash script snippet to `$home/.bashrc`. There is no way (currently) for an upgrade to detect if it's installed or not.
This module doesn't work like a normal installed module (where a directory for it is created).

Perhaps the best way is to detect the current installed state by greping for `RETROPIE PROFILE START` and/or `RETROPIE PROFILE END` in `$home/.bashrc`, pretty much similar to how the removal is actually done:

https://github.com/RetroPie/RetroPie-Setup/blob/2d145c9d98b3e28ba4f60ab6f340a8801b3999c7/scriptmodules/supplementary/bashwelcometweak.sh#L135

Later tonight, when I have access to my retropie setup, I will check all the core packages behaviour to ensure which ones aren't behaving correctly and report back. Thanks!

joolswills, 2018-02-27:
it can't be the same issue - the issue with the menu was due to the update hook marking it as installed because the gamelist was present. This doesn't happen with the bash tweak. Its also a configuration module and should work differently, but I will check (It shouldn't be called at all on updating - it's just something you can manually add or remove)

joolswills, 2018-02-27:
I can confirm the issue - it's a different cause. The "$md_inst" should not be created at all but is present on the image. I will need to do a workaround now to make sure that gets removed (but it shouldn't have been created)

joolswills, 2018-02-27:
It's not created if manually removing `/opt/retropie/supplementary/bashwelcometweak` then re-installing for example.

joolswills, 2018-02-27:
it's to do with the way bashwelcometweak is called for the image created vs calling it manually.

I have adjusted the gui_bashwelcometweak function to resolve this.

joolswills, 2018-02-27:
Thanks for the reports.

hhromic, 2018-02-27:
Very glad to help! and thanks for your prompt answers and fast fixes!

I just tried the latest revision on my instalation and I can confirm both issues were fixed now.
I also tested removing each package in `core` and then upgrading, and I can also confirm that all of them remained uninstalled (worked as expected).

Lastly, apologies for I mistakenly said `bashwelcometweak` was a `core` package, however it actually is a configuration module.

commit c1f736f2 (2018-02-27): retropiemenu - fix menu reappearing after removal - fixes #2320
 * the gamelist.xml was not removed on removal of the module, and this is used to detect an upgrade from older retropie where no $md_inst was created. This caused the module to be marked as installed when running the update hook
 * the delSystem call had the wrong parameter order, causing the es_systems.cfg entry to not be removed