RetroPie/RetroPie-Setup #2321: amiberry.sh build script for Odroid XU4

Tillman-Z, 2018-02-27:
I had to remove all -sdl2 postfixes from the amiberry.sh build script. 
Then it build and installed flawlessly using the retropie-setup.sh.

Best,

Tillman

psyke83, 2018-02-27:
It's caused by this commit: https://github.com/midwan/amiberry/commit/fba61a387ffe784251ea10b2a529667227e7e763#diff-b67911656ef5d18c4ae36cb6741b7965

@midwan: do you plan to add back the -sdl2 postfix, or should I change the RetroPie scriptmodule? Any other changes for other targets planned?

midwan, 2018-02-27:
Sorry about that!
My logic was that we should only keep a postfix for targets that have multiple flavors, to avoid longer filenames for no reason.

Only the RPI targets should really have a postfix, since the rest run in sdl2 exclusively.

Could you please update the script accordingly?

psyke83, 2018-02-27:
Should be fixed now - thanks.

PR #2322 (2018-02-27): amiberry: drop -sdl2 postfix for xu4 target
Fixes #2321.
commit 091fb44a (2018-02-27): amiberry: drop -sdl2 postfix for xu4 target

Fixes #2321.
closed 2018-02-27 by psyke83