RetroPie/RetroPie-Setup #2325: Feature Request: Shipping default configuration for known controller types

hzulla, 2018-02-28:
First of all, thanks for Retropie. It's wonderful!

This issue is a feature request to simplify the controller setup.

Right now, users have to configure the controller settings each time after a fresh install. This is easy for SNES-type controllers, but bothersome for many special settings such as dual shock or analog controllers.

It would be beneficial if Retropie came with a collection of default configurations for known/popular game controller devices, so that these controllers simply work out of the box, with all features activated. If a device such as the original Microsoft XBOX game controller is detected by its USB or bluetooth product ID, Retropie should offer to install the proper driver and activate features like vibration in the PSX emulator.

If a feature like this existed, users could later contribute their configuration files for other controllers, improving the default collection over time.

dankcushions, 2018-02-28:
retroarch already have a bunch of these: https://github.com/libretro/retroarch-joypad-autoconfig/tree/master/udev (however their default layouts aren't always neccesarily what we'd choose).

I like the idea but I have a feeling we've discussed this before and it was decided against with a good reason, but I can't remember the reason :)

hzulla, 2018-02-28:
It was just curious that there are several discussions online about how to fix the "blinking xbox controller" or the "dual shock doesn't work with PS3 controller". People seem to have the same problems with their controller setups over and over.

joolswills, 2018-02-28:
Sorry, but please use the forum as detailed when opening an issue.

Also - we used to do this - it caused many problems. Current method is more flexible.

closed 2018-02-28 by joolswills

joolswills, 2018-02-28:
Many controller issues are due to other factors also. Eg. Initial connecting. Fake / clones. Same ID, different layout. PSU issues.