RetroPie/RetroPie-Setup #2326: I have a way to run ps2 emulation on the raspberry pi or arm based processers

matthewjp, 2018-03-10:
i have recenlty saw a real ps2 emulator app that got taken out of the google play store but is still able to download and crack the apk and its based from ppsspp emulator for android and it was a bit laggy but worked and was wondering can someone put this into a new emulator for retropie? https://www.youtube.com/watch?v=hPXArVob934

matthewjp, 2018-03-10:
https://www.youtube.com/watch?v=WZUnVPhJIcs

matthewjp, 2018-03-10:
the app was called DamonPS2

lilbud, 2018-03-10:
Damon is a lot of BS: https://pcsx2.net/286-the-pcsx2-team-s-statement-regarding-the-damonps2-emulator.html

Besides, you'd have to get Android running on the Pi. Which takes a lot of headroom on the pi. Even if you got this running , it would run _very_ slowly.

joolswills, 2018-03-10:
Please use the forum as directed when opening an issue.

closed 2018-03-10 by joolswills

matthewjp, 2018-03-10:
but would it better better then a low end pc running a ps2 emultor