RetroPie/RetroPie-Setup #2327: Amiga LHA File Support

HoraceAndTheSpider, 2018-03-12:
Apologies in advnace that I am having a bit of a blonde moment in trying to add this code myself, and failing miserably at finding the right places to make the changes, so i will have to revent to making a request...

Amiberry’s development branch has just had a new file support added, LHA. This is commonly used as a single file solution to packaged WHDLoad games, as this format is able to accommodate certain constraints of the Amiga filesystem within a contained archive. (Eg. Issues with certain filenames etc)

The support is fairly simple. Just by adding `-autowhdload=gamefile.lha` on the command line.

For RetroPie to fully take advantage of this (and it was designed with RetroPie in mind) I’d like to suggest that the `es_systems.cfg` has .lha as a supported Amiga filetype and that the following be added to the Amiberry launch script on line 13 (right after the .uae file support)

```
[elif [[ "$rom" == *.lha ]]; then
"$emulator" -autowhdload="$rom" -G]) 
```

Details on how the system is able to have game-specific settings, without using individual Config files, is provided on the Amiberry Wiki Page , and you can expect a large number of game-specifics to be added very soon.

I appreciate that the devs here might want to wait until this change reaches the master branch, but I anticipate that will be quite soon anyway.

Thanks!

psyke83, 2018-03-13:
Does #2328 look OK? We can delay merging until amiberry's master branch catches up. This feature is working on the dev branch? I'd like to compile a build from the dev branch (via a local edit to my script) to make sure everything's ok.

HoraceAndTheSpider, 2018-03-13:
It does look good, thank you @psyke83

I guess that explains why I could not find the script when I searched - I did not realise it was still shared with Uae4arm!

Yes, the feature has been added to the dev version, but you will need to download the full package and not just update the binary as the whdbooter files need including. I have been doing extensive testing to but I made the script changes by hand. 

I also have not created an extensive XML file for the improved game compatibility, but so far I have been pleasantly surprised with how many games work on the default options. I hope I can do that before the next merge with the master branch by it actually isn’t essential. 

There is extra info here if you need it:

https://github.com/midwan/amiberry/wiki/WHDLoad-Auto-booting

Thanks again