RetroPie/RetroPie-Setup #2339: Dolphin fails to load rom/iso

samplethief, 2018-03-19:
Click on any Gamecube rom/iso and fails to launch, returns to rom browser.

Emulator has been installed, updated via the emulationstation experimental package manager menu but the folder Emulationstation calls Dolphin-emu at /opt/retropie/emulators/dolphin is empty...

/dev/shm/runcommand.log says:

`Executing: /opt/retropie/emulators/dolphin/bin/dolphin-emu -b -e "/home/myusername/RetroPie/roms/gc/Mario Kart - Double Dash!! (USA).iso" /opt/retropie/supplementary/runcommand/runcommand.sh: line 1006: /opt/retropie/emulators/dolphin/bin/dolphin-emu: No such file or directory`

Fixed the issue with Dolphin by creating a sym link from /usr/games/dolphin-emu-nogui to /opt/retropie/emulators/dolphin/bin

joolswills, 2018-03-19:
is this your post ? https://retropie.org.uk/forum/topic/16900/dolphin-folder-is-empty

samplethief, 2018-03-19:
It is not @joolswills

samplethief, 2018-03-19:
@joolswills It isnt. But yes looks like the same thing.

joolswills, 2018-03-19:
For future reference, please use the forum first for reports, as detailed when opening an issue. There are more people there to help.

I will need a build log from RetroPie-Setup to look into this as it installs fine here (I don't have any `/usr/games/dolphin-emu-nogui`  - dolphin installs to  `/opt/retropie/emulators/dolphin/bin`)

I also need full details of your set-up including hardware / operating system and version of retropie-setup script you are using.

samplethief, 2018-03-20:
Distributor ID: Ubuntu
Release: 16.04
Codename: xenial
CPU: Intel(R) Core(TM) i3-5010U CPU @ 2.10GHz

Installation Script used: https://github.com/RetroPie/RetroPie-Setup

Dolphin Build Log: https://mega.nz/#!Os0UBDiJ!3OxcB-gAIMSVurDLaqVZYRH5gUNCBVV0svPIwaAfhmE

joolswills, 2018-03-20:
The build fails - it should actually not run the configure, but the module needs some extra code to know that the build has failed to do this. I will do that, but it may just be due to compiler versions on Ubuntu 16.04 as to why it's failing. I built fine on Ubuntu 18.04 (beta). I probably won't get around to looking at this any time soon, so it might be worth trying a newer Ubuntu or waiting till the new LTS is out.

samplethief, 2018-03-20:
@joolswills Yeah no problem I was just bringing it to your attention, As I say, I have it working fine using a symbolic link to the actual location.

joolswills, 2018-03-20:
You must have installed another copy for it to be installed there - via Ubuntu packages I assume ?

samplethief, 2018-03-20:
ah yes, I installed with apt-get install dolphin-emu. Ill add that to the description incase anybody else comes here looking for solutions in the mean time.

joolswills, 2018-03-20:
I confirmed the issue in a VM. I have worked around the problem by forcing an older version to be built on Ubuntu 16.04.

closed 2018-03-20 by joolswills

samplethief, 2018-03-20:
Nice work thanks

commit eb9d1070 (2018-03-20): dolphin - latest HEAD doesn't build on Ubuntu 16.04 - use 5.0 branch for older GCC - fixes #2339