RetroPie/RetroPie-Setup #2347: lr-ppsspp build script asks for username and password

drwhoelse, 2018-03-27:
Below is copy-n-paste of terminal box for building lr-ppsspp...

= = = = = = = = = = = = = = = = = = = = =
Getting sources for 'lr-ppsspp' : PlayStation Portable emu - PPSSPP port for libretro
= = = = = = = = = = = = = = = = = = = = =

git clone --recursive --depth 1 "https://github.com/libretro/libretro-ppsspp.git" "/home/ubu/RetroPie-Setup/tmp/build/lr-ppsspp"
Cloning into '/home/ubu/RetroPie-Setup/tmp/build/lr-ppsspp'...
Username for 'https://github.com': 
Password for 'https://github.com':
-------

After I entered blank username and password, I get error 128 message.

joolswills, 2018-03-28:
upstream have removed the repo as they have a new ppsspp core. We don't yet support that yet. I may be able to restore a fork of the old repo if I have a local checkout, or else we will have to disable the module. You can use the standalone PPSSPP emulator.

joolswills, 2018-03-28:
actually we have a fork already, so I can just use that for now.

joolswills, 2018-03-28:
This is fixed in fcd8367616283df50e444d399101716265dfa9dd but please provide more information next time including a full log to help us (This issue was specific to non rpi platforms).

closed 2018-03-28 by joolswills