RetroPie/RetroPie-Setup #2349: Wiki examples incomplete?

anohren, 2018-03-28:
It appears that [the examples on this page](https://github.com/retropie/RetroPie-Setup/wiki/Building-RetroPie-Archives) are incomplete. It's listing the same command three times, i.e. `sudo ./retropie_packages.sh builder module <MODULEID/MODULENAME>`.

joolswills, 2018-03-28:
It's not complete and looks like c&p fail - but this doesn't warrant an issue opening here. Please use the forum for discussion of the wiki. I may just remove that page - those functions are primarily for distribution and so are not useful for end users.

closed 2018-03-28 by joolswills

anohren, 2018-03-28:
The issue template did not discourage it. I am not aware of any forum. As you please.

joolswills, 2018-03-28:
You didn't read the issue template then

>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

>Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

>For ideas and enhancements, please also post to the forum first, to see if it is something we would consider and to see if other users would find it useful.

>Please remove this text before posting.

anohren, 2018-03-28:
This is not a request for help or support. It is not a bug that can be solved with configuration changes. It is not an idea, nor a request for enhancement.

I apparently read it, since I removed the text. There's no need to make baseless accusations.

joolswills, 2018-03-28:
You said you were not aware of any forum so I obviously assumed you hadn't read it.