RetroPie/RetroPie-Setup #2384: glGetError 0x505

actraiser30, 2018-04-25:
Hi, i installed latest Retropie 4.4 on  a Raspberry pi 3 and putted some games, and the main theme looks terrible after a while  I get the white screen and font corruption by quitting to command line with F4 i get the glGetError 0x505 
i tried several themes the same problem happening in all of them.
i have fully updated Retropie with latest setup-script
i had set vram to 80-100-130 still the problem occurs.
any solution?
Thanks

joolswills, 2018-04-25:
You didn't follow the instructions when opening an issue:

>Please start a topic on the RetroPie forum before opening an issue - https://retropie.org.uk/forum/

>This includes edit suggestions for the wiki. There are more people to help on the forum.

>Once a problem has been verified on the forum, an issue can be opened here.

closed 2018-04-25 by joolswills

actraiser30, 2018-04-25:
at least you can fix that bug that goes on forever instead of sending me somewhere else.

joolswills, 2018-04-25:
You are the first to mention it. If verified by others we will look into it. Please follow our guidance thanks.

mydmitry, 2018-06-03:
Hi guys.
Go to Retropie configuration (press A at main screen on Retropie)
then
RASPI-CONGIG -> 7 Advanced Options -> A3 Memory Split

If you will see there for example 64mb so it's <80mb you set for VRAM retropie.
For me it was changed from 256mb to 64mb as I thinks with last update.

Hope it helps.

brianolsen87, 2018-12-10:
Some more things to add to mydmitry's post. I didn't have access to any of the gui options via emulation station nor RetroPie to change my VRAM, so I ran this from the terminal.

`sudo raspi-config`

Then selected the Advanced Options -> A3 Memory Split as mydmitry suggested.

Another thing to note is that my VRAM was actually 32mb and I needed to increase it to 128mb before the issue was resolved. So decreasing it may not always be the answer. You'll need to calibrate it based on the software versions/pi versions you're using.