RetroPie/RetroPie-Setup #2385: lr-ppsspp (and ppsspp) issue after Retropie 4.4 update

cgreeney, 2018-04-25:
Pi Model or other hardware: Raspberry Pi 3 B+
Power Supply used: 2.5A Power Supply from canakit
RetroPie Version Used: 4.4
Built From: Pre made SD Image on RetroPie website
USB Devices connected: keyboard, controller
Controller used: PS3 Generic controller

I go to run lr-ppsspp after the update and get the following message (for a PSP Minis game which uses lr-ppsspp) in /dev/shm/runcommand.log. When you initialize whatever rom you want to play, it just kicks you back to ES main menu.  Output is as follows:
```
/opt/retropie/emulators/retroarch/bin/retroarch`` -L /opt/retropie/libretrocores/lr-ppsspp/ppsspp_libretro.so --config /opt/retropie/configs/pspminis/retroarch.cfg "/home/pi/RetroPie/roms/pspminis/1000 Tiny Claws (USA).cso" --appendconfig /dev/shm/retroarch.cfg

glGetError 0x500

/opt/retropie/emulators/retroarch/bin/retroarch: symbol lookup error: /opt/retropie/emulators/retroarch/bin/retroarch: undefined symbol:
```

cgreeney, 2018-04-26:
Thanks for your prompt reply.  I did start a topic on the forums without answer.  Have you possibly tried restarting the system to see if it still works as well?  It worked once and after a reboot it was non-existent.

joolswills, 2018-04-26:
Maybe you have some sd card corruption. Works fine here after a reboot. You didn't wait that long after posting at the forum to post here btw. 

Please use the forum for further help.

cgreeney, 2018-04-26:
Everything else works fine other than lr-ppsspp after two re-images and trying other images from the past doesn't yield any issues in regards to card corruption.  I'll start again from the drawing board.

Will do.  Thanks!

SA200916, 2018-09-01:
I have this same problem but reinstalling the emulators and using a different card doesn't change the result. I've tried multiple display and rendering modes, did anyone find a solution? or have a suggested troubleshooting path i might try? Thanks in advance