RetroPie/RetroPie-Setup #2397: finalizing the mame2003-plus build script

markwkidd, 2018-05-04:
@joolswills I'm taking your suggestion and not submitting a PR to RetroPie-Setup.

I agree that an Issue will be a better place to make sure we're on the same page about the mame2003-plus install script.

### The one big issue now
twinaphex had to rename the mame2003-plus binary in the Makefile again. This was due to name conflicts in the automated buildbot system over at libretro.

I apologize that I jumped in to try to get this set up in RetroPie just at a time when it turned out we would be making these name changes a few times. 

### Minor issues

If you do get time to update the build scripts again, could I make a suggestion in terms of cleanup? 

The scripts are currently making two folders that mame2003 and mame2003-plus don't use: `inp` and `snap`

In fact, as of the last couple of months, both versions of mame2003 generate the remaining folders on their own : `cfg`, `ctrlr`, `diff`, `hi`, `memcard`, `nvram` (and `artwork` for -plus). There is no longer a need for RetroPie-Setup to create the folders if you want to remove code as well.

Thanks for your work on RetroPie. I hope this makes sense, and just let me know if it doesn't or if I can explain anything else that's been going on in mame2003 land.

joolswills, 2018-05-04:
ok thanks. Will sort. Will tag you in a PR to check if the changes seem logical.

markwkidd, 2018-05-04:
> It might still be useful for our script to make some folders, as users can
> see them / copy stuff into them before launching the binary.

Thank you.

That makes sense about generating any folders the user might want to
prepopulate. In that spirit can I suggest then that the `artwork` folder be
added to the list of pregenerated folders?

joolswills, 2018-05-05:
is it just the library name that has changed ? the mame2003-plus folder is still mame2003-plus not mame2003_plus ?

markwkidd, 2018-05-05:
Yes, my understanding is that there would be no need to rename the folder.
As far as I know it is just the changed filename of the compiled binary
that causes RetroPie-Setup to stop.

joolswills, 2018-05-05:
https://github.com/RetroPie/RetroPie-Setup/pull/2398

needs testing.

markwkidd, 2018-05-05:
ah. I don't know what exactly required the underscore but it was pretty
clear from the brief discussion that the change was unavoidable.

grant2258, 2018-05-06:
is just the target.so that changed @joolswills. I done a temp fix not a pretty one but just to get it compiling for users I didnt want to mess any other stuff up. I just copied the original mame function and added a copy command will test the script you done and get back to you

function build_lr-mame2003-plus() {
    rpSwap on 750
    make clean
    local params=()
    isPlatform "arm" && params+=("ARM=1")
    make ARCH="$CFLAGS" "${params[@]}"
    rpSwap off
    cp -f mame2003_plus_libretro.so  mame2003-plus_libretro.so
    md_ret_require="$md_build/$(_get_name_lr-mame2003)_libretro.so"
}

grant2258, 2018-05-06:
the build script works as it should its all good there is only one other concern I have, Its in regards to creating the directorys. This will change where it needs to go dependent on the directory structure. Retroach is looking for the following for savings paths. 

/libretro savefile dir/mame2003-plus/cfg/
/libretro savefile dir/mame2003-plus/diff/
/libretro savefile dir/mame2003-plus/hi/
/libretro savefile dir/mame2003-plus/memcard/
/libretro savefile dir/mame2003-plus/nvram/

this is the latest in the documentation @markwkidd  can you verify this information is still correct?

if i have my roms in 
/home/pi/Retropie/arcade savefile will be /home/pi/Retropie/arcade/mame2003-plus

if i have my roms in 
/home/pi/Retropie/arcade/mame2003-plus-split/ /home/pi/Retropie/arcade/mame2003-plus-split/mame2003-plus

What im trying to say is we need to set the savefile path in retroach else it will use the content directory.

joolswills, 2018-05-07:
We use the content directory for saves for everything else. I don't see a problem with that.

grant2258, 2018-05-07:
that no problem if that the way you guys want it. Some user will have issues though as libretto requires you to use the savefile path set in the retroarch.cfg else it will use the content dir. It will only cause issue if people have roms in sub folders anyway.

dankcushions, 2018-05-07:
can you explain what the issues are? is it that if you have a rom in `/roms/mame2003-plus/foo/`, it will try and look for stuff in `/roms/mame2003-plus/foo/hi/` etc, and that won't work because they won't be there?

grant2258, 2018-05-07:
mame will create the dirs its needs anyway the problem arises when you need to copy content from the metadata to one of these dirs on install the nvram fix for example.

if savefile is set

MAME will look in %savefilepath%/mame2003/hi
MAME+ will look in %savefilepath%/mame2003-plus/hi
 
regardless of what folder you roms are in is will always use this dir

in the case of save file is not set eg  /roms/mame2003-plus/foo/

it will look for the data in

/roms/mame2003-plus/foo/mame2003/hi
/roms/mame2003-plus/foo/mame2003-plus/hi

mame will create the folders and such but you will loose any data that was previously in and directory if its not moved

dankcushions, 2018-05-07:
> in the case of save file is not set eg /roms/mame2003-plus/foo/
> it will look for the data in
> /roms/mame2003-plus/foo/mame2003/hi
> /roms/mame2003-plus/foo/mame2003-plus/hi
> mame will create the folders and such but you will loose any data that was previously in and directory if its not moved

ok, but specifically what's the issue? say i launch a game in a subfolder: `/roms/mame2003-plus/foo/foo.zip`, i then do a hiscore: `/roms/mame2003-plus/foo/mame2003/hi/foo.hi` (mame2003 creates the folder structure), and then save a state: `/roms/mame2003-plus/foo/foo.state` (retroarch uses content dir as save dir).

does this work? if not, what gets lost?

grant2258, 2018-05-07:
@dankcushions nothing gets lost when mame is running say you rename you rom dir all is good the file is in there. The problem is on install lets look at the nvram fixes for example

if your roms are in foo

when installing it will install to 
/roms/mame2003-plus/nvram

instead of
/roms/foo/mame2003-plus/nvram

dankcushions, 2018-05-07:
right, so the games that need these nvrams won't load because retroarch would be looking in the sub directory, not the main directory (where they are) - got it, thanks!

perhaps this is a good enough reason to add 'savefile_dir' = /blah/mame-libretro/ via iniSet, eg like https://github.com/RetroPie/RetroPie-Setup/blob/60b03ffce8094cebe404ae78e8ce802d7bde5d57/scriptmodules/libretrocores/lr-bluemsx.sh#L47 - you'd have to do it for both arcade and mame-libretro retroarch.cfgs

...but, i also think that subfolders can cause other issues with scraping, ES, things like that, so maybe aren't recommended anyway. not sure if this use case is worth handling.

joolswills, 2018-05-07:
I don't want to add a config for just this one for consistency. If people want to use subfolders for mame (which is unusual imho), they can always add it (and it can be documented on the wiki if needed).

Perhaps for games that need a default nvram (seems like a hack/workaround to me), it would be better to just bake them in, or have a "default" stored in BIOS location.

markwkidd, 2018-05-07:
For what it's worth I'm going to try to use bin2c to bake the patches into the core like with rungun so that we don't need to distribute default nvram files.

Also for what it's worth, hack or no default nvrams are how MAME has solved this for some of these games -- at least through MAME 0.139. Later versions of MAME like 0.139 include the nvram in the game romset zip, but MAME 0.78 predates that approach.

derail over.

grant2258, 2018-05-07:
@joolswills  how does retropie handle mutiple romsets for mame ie mame2000 mame2003 and mame2010  which all use different sets with no subdirs?

joolswills, 2018-05-07:
By default you have to choose. Or you can mix sets with some manual work switching emulators etc. I didn't want a rom folder per set but we have a few folders people can choose from etc.

joolswills, 2018-05-07:
@markwkidd bin2c sounds like a good solution.

grant2258, 2018-05-07:
i just used sub folders the way it was setup so my sets worked right I still use it this way and have no problems again if users are aware they shouldnt be making sub folders its up to them.  Its a lot of work to make sets work with multiple emulators ie the fba/mame neogeo bios issue

closed 2020-11-10 by markwkidd