RetroPie/RetroPie-Setup #2420: Logitech F710 buttons working as hotkeys in lr-pcsx-rearmed, and some fixed VICE problem

Vendi76, 2018-06-06:
Model: raspberry pi 3B+
retropie 4.4

I'm using a logitech f710. In Emustation, it's OK. 
I had some issues under VICE, the left analog joy "up" function didn't worked, but luckily found out, that somehow when VICE saved joystick settings, it mysteriously saved a "broken" line for the "up" direction, managed to correct it manually.
When I'm running psx games under lr-pcsx-rearmed, all the buttons work like hotkeys, pressing "x" starts RGUI menu, "start" exits etc.

Vendi76, 2018-06-06:
I was a lame :) followed retroarch controller setup instructions videos from youtube, and with "autosave on exit" enabled i've messed up the whole thing. It works just perfect, if i only config in emulatorstation, and no additional messing in retroarch.

closed 2018-06-06 by Vendi76

joolswills, 2018-06-06:
Please also use the forum for support in the future as detailed when opening an issue. thanks.