RetroPie/RetroPie-Setup #2422: ppsspp 1.6.3 build script. ETA

gerble1000, 2018-06-08:
Please start a topic on the RetroPie forum before opening an issue - https://retropie.org.uk/forum/

This includes edit suggestions for the wiki. There are more people to help on the forum.

Once a problem has been verified on the forum, an issue can be opened here.

Please remove this text before posting.

gerble1000, 2018-06-08:
i cannot seem to build the ppsspp emulator on my pie 3.
im looking forward to a retropie build script :)

psyke83, 2018-06-10:
Closing. In future, be sure to follow the advice in the issue template (that you neglected to delete & replace when opening this ticket). 

There's an active topic already: https://retropie.org.uk/forum/topic/18116/ppsspp-emulator-update-v1-6-3

closed 2018-06-10 by psyke83