RetroPie/RetroPie-Setup #2436: Disable DS3/DS4 motion controls by default?

rsn8887, 2018-06-26:
DS3/DS4 motion controls are detected as separate controller and interfere with some emulators such as AdvanceMame.

There's a discussion about this in the RetroPie forum, and I found a simple solution via a udev rule as detailed here:
https://retropie.org.uk/forum/post/152584

Can this solution be incorporated into the next RetroPie release?

psyke83, 2018-06-28:
I've already patched customhidsony to disable the motion sensor node via the driver, but I haven't tried this approach. Does it solve the issue of the motion controller occupying a controller slot in userspace? To illustrate, if I connect two DS3 controllers with the unpatched hid-sony driver, the kernel assigns the first pad to player 1 and the second to player 3 (with player 2/4 assigned to the motion sensor node).

psyke83, 2018-06-29:
I'm uncertain if there's a userspace tool to check definitively, but you can observe it happening by the actual controller LEDs (first pad lights LED 1, second lights LED 3), and the second pad doesn't function as player 2 in retroarch.

I'll check this out when I have some free time, but in the meantime, this is the patch to disable sensors that's applied via customhidsony: https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/supplementary/customhidsony/0001-hidsony-nomotionsensors.diff

I isolated this change to sixaxis controllers, as I don't have the DS4 controllers and thus wasn't sure if it had the same issue.

rsn8887, 2018-06-29:
DS4 controllers have the same issue. 

DS4 don't have those player LEDs. However, I see that with my solution, with a single DS4 controller connected, the controller is registered in AdvanceMame as "js2." This sounds like the player numbers are still messed up, since it should be "js1" shouldn't it?

psyke83, 2018-06-29:
It's an optional install, which provides a very minor update to the hid-sony driver (backported from kernel 4.15) mainly to fix an infinite rumble via USB issue with Shanwan controllers.

I'll take a look again to see if we can achieve the same goal via something similar to your method or another sysfs userspace method. If not, I'll adjust the patch so that it also disables the sensor node on DS4 controllers too.

I've already got another PR pending that aims to help work around hid-sony driver quirks, including an additional patch to disable the digital trigger buttons in the driver: https://github.com/RetroPie/RetroPie-Setup/pull/2263

The overall aim is to replace the ps3controller userspace wrapper in favour of using the native BlueZ plugin, but the changes here can most likely also benefit DS4 pads. I might need to add some udev entries so that DS4 pads can also avail of the idle timeout function, though.

rsn8887, 2018-06-29:
Never mind forget what I said. 

My user rule seems to work perfectly I have player 1 and 2 they register correctly as player 1 and 2, also in AdvanceMAME it works as it should (remapping shows j and j2 correctly.)

I must have remembered it wrong last time

rsn8887, 2018-11-22:
Can this be implemented? I appreciate the other effort of patching a separate driver. 

But personally I don’t like the idea or see the reason of an optional driver install if the default system driver works fine, with just a simple added udev rule to mute the motion controls.

In addition, this approach will keep Retropie future proof since I expect the system driver to have attention from the greater community...

psyke83, 2018-11-22:
The ```hid-sony``` driver in kernel 4.14 (Raspbian stretch) is broken for third-party controllers: Shanwan have an eternal USB rumble bug, and Gasia won't pair via BT (which isn't fixed in any upstream kernel release yet).

I had forgotten about your udev rule; while it may at least allow the motion control patch to be removed, we still need to patch the driver for the aforementioned reasons. I'll take a look soon and update my PR with your udev rule if it works OK (i.e. doesn't mess up the LED assignments on DS3 controllers).

rsn8887, 2018-11-22:
Thanks. Maybe two udev rules can be added, one for your new patched driver and one for the default driver? This way people who use first party controllers don’t need to do anything, and people who have third party controllers can install your patched driver.

It makes sense to me that the system driver supports only first party controllers, especially if the knock-offs are (apparently) not 100% compatible. Since most users have first party controllers, the system driver will be fine for them, and requires zero additional work apart from this time.

psyke83, 2018-11-22:
No, the patched driver is not for for third-party controller support. I'm not certain what the situation is with DS4 controllers, but the default ```hid-sony``` driver *on its own* is not a viable replacement for the ```ps3controller``` (```sixad```) driver on a DS3. Reasons include:

* the default axis configuration has jitter (i.e., the resting position of the analog sticks produces a constant stream of evdev events).
* the sixaxis plugin of BlueZ doesn't support the system Bluetooth IdleTimeout setting. This means that once the controller is paired, it will stay paired until the user forcibly disconnects the pad (holding the PS button for 10 seconds), the system turns off or when the battery runs flat. This is totally unacceptable for a lot of people that keep their Pi on 24/7.
* the L2 & R2 buttons map to a digital button as well as virtual axes (to report button pressure). This has the effect of confusing any applications that require joystick button mapping. Kodi has a special feature to workaround this, but EmulationStation will not be able to map these buttons correctly (as it will interpret two separate button events for a single press of those buttons).

My PR is going to add an optional driver module that's installed instead of ```ps3controller```, with the aim of addressing all of these issues. The patched driver will therefore be optional, so I don't see the point in complaining about something that's not going to be foisted on everyone.

I've added your udev rule to my sixaxis scriptmodule as it's definitely useful to mitigate one of the issues with the driver, but that package will also install the patched driver alongside it. The RetroPie image only includes a few generic udev rules to relax permissions, IIRC, so I'm not sure if adding such a controller-specific workaround by default is wise, especially when the driver already suffers from plenty of other issues by default.

rsn8887, 2018-11-23:
Understood. I wasn’t aware of the other issues with the default driver. I didn’t mean to complain about your driver, rather about why the system driver was considered broken. Now it is clear.

psyke83, 2019-02-28:
The `sixaxis` scriptmodule is now merged into master; this will install the udev rule to hide motion controls and make sure that applications don't erroneously identify the controller as a keyboard (which is a bug that's already fixed in later upstream versions of systemd).

closed 2019-02-28 by rsn8887

joolswills, 2019-02-28:
Thanks @psyke83 :-)

ryanw4130, 2019-03-26:
Thanks for the script instructions! I also modified it to disable the touchpad.
I am running 2 dual shock 4 controllers on Retropie v4.4, with on-board Bluetooth disabled. I have them paired to a Bluetooth 4.0 USB adapter, and I noticed that if you press the touchpad in megadrive games, the pi would freeze. This got real annoying. So I copied your script and added another line to disable to DS4 touchpads. This did the trick, or at least I tested it on some megadrive games, and it no longer freezes. Going to configure the controllers in reicast, and the touchpad no longer shows up as a input.

Maybe someone else can find this useful. Here is the what i used.

#disable DS3/DS4 motion controls
SUBSYSTEM=="input", ATTRS{name}=="*Motion Sensors", RUN+="/bin/rm %E{DEVNAME}", ENV{ID_INPUT_JOYSTICK}=""
SUBSYSTEM=="input", ATTRS{name}=="*Touchpad", RUN+="/bin/rm %E{DEVNAME}", ENV{ID_INPUT_JOYSTICK}=""