RetroPie/RetroPie-Setup #2440: Scraper stops after scanning, doesn't update gamelist

jdruding, 2018-07-01:
I have Retropie 4.3 on a Raspberry Pi 3. I had previously run sselph's scraper successfully, but have since added some new games that I wanted to scrape. To scrape these newly added games, I exited ES, updated the scraper via the menu option, set it to "append" and starting running. The first time I did all systems. It got the end and just stayed at the command line, with the cursor flashing, and never gave me the message that it completed. After a while, I restarted and tried again, but only for arcade this time. Same issue, it got to the last game and just stopped and never gave me a completion message. If you reboot and go back in to emulation station, you can see that it hasn't updated the gamelist with the games that should have been added.

joolswills, 2018-07-01:
Please use the forum first as directed when opening an issue.

closed 2018-07-01 by joolswills

jdruding, 2018-07-01:
Did that...didn't get an answer:

https://retropie.org.uk/forum/topic/18262/sselph-scraper-stops/2

andrewjameshough, 2018-07-01:
You did though? Have you updated to stretch now?

jdruding, 2018-07-01:
No, I have not updated because I would not consider a sentence containing "I don't know" to be definitive enough direction to make such a drastic change without further corroboration.  I don't see any other references to this being the solution and no one else commented.  Are you confirming that this is the issue?

andrewjameshough, 2018-07-02:
Luckily the change isn't too drastic it's relatively easy to upgrade your dist (if you do a quick search there are quite a few good guides). All I can do is confirm I don't have this issue and I'm running the latest build on stretch which implies this may fix the issues you're seeing.