RetroPie/RetroPie-Setup #2443: Unable to compile dolphin. QT 5.9 required

raidensnake, 2018-07-21:
Recently Dolphin won't compile as it requires version 5.9 or better. This isn't supported on Ubuntu 16.04 or Debian Stretch as the packages are older.

raidensnake, 2018-07-22:
`CMake Error at Source/Core/DolphinQt/CMakeLists.txt:1 (find_package):
  Could not find a configuration file for package "Qt5" that is compatible
  with requested version "5.9".

  The following configuration files were considered but not accepted:

    /usr/lib/x86_64-linux-gnu/cmake/Qt5/Qt5Config.cmake, version: 5.7.1

-- Configuring incomplete, errors occurred!`

joolswills, 2018-07-22:
>Please start a topic on the RetroPie forum before opening an issue - https://retropie.org.uk/forum/

>This includes edit suggestions for the wiki. There are more people to help on the forum.

>Once a problem has been verified on the forum, an issue can be opened here.

If Dolphin requires a newer Qt5 then there is nothing we can do - you will have to upgrade to 18.04

closed 2018-07-22 by joolswills