RetroPie/RetroPie-Setup #2449: added iortcw to experimental

cesarpuig, 2018-08-03:
Hi. first of all I forgive because I think this should be a pull request, but I do not know how to do it.

I have installed iortcw-master 1.51c in a Raspberry pi 3 B, I have created a script to launch it from emulationstation:

#! / bin / bash
"/opt/retropie/supplementary/runcommand/runcommand.sh" 0 _PORT_ "iortcw" ""

and a configuration file:

iortcw = "LD_LIBRARY_PATH = lib /opt/retropie/ports/iortcw/iowolfsp.arm"
default = "iortcw"

I work perfect.

Is there the possibility of adding it to experimental?

Thank you very much.

actraiser30, 2018-09-13:
Hello great work do you have info about how to install/build it to retropie?

cesarpuig, 2019-12-23:
Hi,
There is a thread in the forum of retropie about it, and there is a scriptmodule to install it.

https://retropie.org.uk/forum/topic/22554/return-to-castle-wolfenstein-iortcw-source-port

I close the issue

closed 2019-12-23 by cesarpuig