RetroPie/RetroPie-Setup #2475: PS3 controller - does not pair with bluetooth

leosmeira, 2018-08-30:
Tested with: 
- RetroPie 4.4.2 (installed manually with 'git clone')
- Raspbian 4.14.62-v7+
- Clone Gasia and Clone shanwan
- with customhidsony or with not customhidsony

I connect the PS3 control with usb cable and then disconnect, it does not pair via bluetooth.

If I run "sixad-helper sixpair" manually, the connection via bluetooth is performed perfectly.

I think this error is in "/etc/udev/rules.d/99-sixad.rules"

andrewjameshough, 2018-09-13:
I think there maybe a wider issue with the pi-bluetooth driver here.. you have to install the earlier one and that seems to do the trick for me. my sfc30 pro wouldn't appear at all with the current driver.

GrieverV, 2018-09-19:
Had the same issue with my Sony DS3 and was able to workaround it by resetting bluetooth and manually pairing, trusting and connecting. The following steps are generic as possible except for which bluetooth device to reset (`hci0`).

```
# hciconfig hci0 reset
# service bluetooth restart
# bluetoothctl
    power on
    devices
    scan on
    agent on
    pair XX:XX:XX:XX:XX:XX
    trust XX:XX:XX:XX:XX:XX
    connect XX:XX:XX:XX:XX:XX
```

Replace XX:XX:XX:XX:XX:XX with the game controller's MAC address. Your device should have been listed when you ran `devices` or `scan on`.

noMICROSOFTbuhtz, 2018-10-04:
Thanks for your detailed description. But I am not using a Pi. I am using a Netbook with Atom (Single Core) CPU.

GrieverV, 2018-10-04:
Ah, I misunderstood. What exactly isn't working? What's the output?

Make sure you have `bluez` and `bluez-tools` installed. You could also try [sixpair](https://help.ubuntu.com/community/Sixaxis#USB_Pairing). sixpair requires `libusb-dev` and `libusb-0.1-4`

noMICROSOFTbuhtz, 2018-10-06:
Ok, lets see if I can get this together... Last night I did a full update of RetroPie.

RetroPie Version: 4.4.2 (last commit 472b0dd)
EmulationStation: 2.7.6RP
Controller used: Original (I think so) PS3 Controllor
Samsung N220 Netbook (Atom)
OS Debian stable: 4.9.0-8-amd64 #1 SMP Debian 4.9.110-3+deb9u5 (2018-09-30) x86_64 GNU/Linux

The packages you mentioned are installed
```
$ apt-cache policy bluez
bluez:
  Installiert:           5.43-2+deb9u1
  Installationskandidat: 5.43-2+deb9u1
  Versionstabelle:
 *** 5.43-2+deb9u1 500
        500 http://ftp.de.debian.org/debian stretch/main amd64 Packages
        500 http://security.debian.org/debian-security stretch/updates/main amd64 Packages
        100 /var/lib/dpkg/status
$ apt-cache policy bluez-tools
bluez-tools:
  Installiert:           0.2.0~20140808-5+b2
  Installationskandidat: 0.2.0~20140808-5+b2
  Versionstabelle:
 *** 0.2.0~20140808-5+b2 500
        500 http://ftp.de.debian.org/debian stretch/main amd64 Packages
        100 /var/lib/dpkg/status
$ apt-cache policy libusb-dev
libusb-dev:
  Installiert:           2:0.1.12-30
  Installationskandidat: 2:0.1.12-30
  Versionstabelle:
 *** 2:0.1.12-30 500
        500 http://ftp.de.debian.org/debian stretch/main amd64 Packages
        100 /var/lib/dpkg/status
$ apt-cache policy libusb-0.1-4
libusb-0.1-4:
  Installiert:           2:0.1.12-30
  Installationskandidat: 2:0.1.12-30
  Versionstabelle:
 *** 2:0.1.12-30 500
        500 http://ftp.de.debian.org/debian stretch/main amd64 Packages
        100 /var/lib/dpkg/status
```

My bluetooth controller
```
$ hcitool dev
Devices:
	hci0	00:26:B6:B1:69:2B

```

The **Configure Input** in the EmulationStation doesn't see a game pad via Bluetooth. Only via USB the PS3-controller is detected.

GrieverV, 2018-10-06:
Only the sixpair part needed to be following.

I attempted to reproduce the issue by pairing my controller with another device, removing the device from the RetroPie device and resetting the bt controller, but couldn't reproduce the initial issue (device not found from RetroPie setup). I did confirm that the device still fails to pair through setup.

I can confirm that the PS3 controller is a bit finicky and needs to be unplugged while running bluetoothctl to appear in the device list. Simply unplugging and replugging after `agent on` and then running `devices` again worked for me.

Your device should appear something like this:

    [bluetooth]# devices
    [NEW] Device XX:XX:XX:XX:XX:XX PLAYSTATION(R)3 Controller
    [CHG] Device XX:XX:XX:XX:XX:XX Class: 0x000508
    [CHG] Device XX:XX:XX:XX:XX:XX Icon: input-gaming
    [CHG] Device XX:XX:XX:XX:XX:XX Connected: yes
    [CHG] Device XX:XX:XX:XX:XX:XX Modalias: usb:v054Cp0268d0100
    [CHG] Device XX:XX:XX:XX:XX:XX UUIDs: 00000000-0000-0000-0000-00ffffffffff
    [CHG] Device XX:XX:XX:XX:XX:XX UUIDs: 00000000-0000-0000-0000-00ffffffffff
    [CHG] Device XX:XX:XX:XX:XX:XX ServicesResolved: yes

noMICROSOFTbuhtz, 2018-10-07:
I am a bit confused about your _unplugged and replug_ thing. I don't understand why and when I have to plug the  PS3-controller via usb and when not.

In the bash output below you see the interaction via bluetooth. The controller was not connected via usb cable.

At the beginning you see that `bluetoothctl` did not start.
```
user@MONSTER:~$ sudo bluetoothctl
[sudo] Passwort für user: 
Waiting to connect to bluetoothd...
user@MONSTER:~$ sudo service bluetoothd start
Failed to start bluetoothd.service: Unit bluetoothd.service not found.
user@MONSTER:~$ sudo service bluetooth start
user@MONSTER:~$ sudo bluetoothctl
[NEW] Controller 00:26:B6:B1:69:2B MONSTER [default]
[NEW] Device 00:26:5C:21:D3:00 00-26-5C-21-D3-00
[bluetooth]# agent on
Agent registered
[bluetooth]# scan
Missing on/off argument
[bluetooth]# scan on
Discovery started
[CHG] Controller 00:26:B6:B1:69:2B Discovering: yes
[CHG] Device 00:26:5C:21:D3:00 Connected: yes
[00-26-5C-21-D3-00]# info
Device 00:26:5C:21:D3:00
	Alias: 00-26-5C-21-D3-00
	Class: 0x000540
	Icon: input-keyboard
	Paired: no
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
[00-26-5C-21-D3-00]# trust
Changing  trust succeeded
[00-26-5C-21-D3-00]# connect
Missing device address argument
[00-26-5C-21-D3-00]# info
Device 00:26:5C:21:D3:00
	Alias: 00-26-5C-21-D3-00
	Class: 0x000540
	Icon: input-keyboard
	Paired: no
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
[00-26-5C-21-D3-00]# pair
Attempting to pair with 
[agent] PIN code: 445954
Failed to pair: org.blue
```

I don't see what I can do here. I am even not sure if the `00-26-5C-21-D3-00` is the ps3-controller or not. I am sure here are lot more bluetooth devices around at my neighbors. And if it is the ps3-controller why does it ask for a pin code?

GrieverV, 2018-10-07:
`pair`, `trust` and `connect` require the device's bluetooth address as an argument.

All you have to do is unplug and re-plug the controller through USB after running all the commands up to `agent on` in [my original reply](https://github.com/RetroPie/RetroPie-Setup/issues/2475#issuecomment-422702521) and then run `devices` again. After running `devices` your PS3 controller should appear as shown above with a unique bluetooth address which you'll use with `pair`, `trust` and `connect`.

You can view more detailed pairing instructions on the [ArchLinux wiki](https://wiki.archlinux.org/index.php/bluetooth#Pairing).

noMICROSOFTbuhtz, 2018-10-09:
It is a bit clearer now. I am still confused about the PIN code.

I connected the controller via USB cabel. Then...
```
$ sudo bluetoothctl
[NEW] Controller 00:26:B6:B1:69:2B MONSTER [default]
[NEW] Device 00:26:5C:21:D3:00 00-26-5C-21-D3-00
[bluetooth]# power on
Changing power on succeeded
[bluetooth]# devices
Device 00:26:5C:21:D3:00 00-26-5C-21-D3-00
[bluetooth]# scan on
Discovery started
[CHG] Controller 00:26:B6:B1:69:2B Discovering: yes
[bluetooth]# agent on
Agent registered
```

Unplugged an re-plugged

```
[CHG] Device 00:26:5C:21:D3:00 Connected: yes
[00-26-5C-21-D3-00]# pair 00:26:5C:21:D3:00
Attempting to pair with 00:26:5C:21:D3:00
[agent] PIN code: 966936
[00-26-5C-21-D3-00]# trust 00:26:5C:21:D3:00
Changing 00:26:5C:21:D3:00 trust succeeded
[00-26-5C-21-D3-00]# connect 00:26:5C:21:D3:00
Attempting to connect to 00:26:5C:21:D3:00
Failed to pair: org.bluez.Error.AuthenticationFailed
[00-26-5C-21-D3-00]# devices
Device 00:26:5C:21:D3:00 00-26-5C-21-D3-00
[CHG] Device 00:26:5C:21:D3:00 ServicesResolved: yes
Failed to connect: org.bluez.Error.Failed
[00-26-5C-21-D3-00]#
```

psyke83, 2018-11-18:
#2263 will be necessary for Shanwan and Gasia controllers to work, so more testing of this PR by people with third-party controllers would be helpful.

With the ```sixaxis``` module of the PR installed, the pairing procedure will be fairly straightforward:
* Enter RetroPie's Bluetooth menu
* Select ```Register and Connect to Bluetooth Device```
* The Searching text box will include this message:
```DualShock registration: wait 3 seconds, then unplug and replug the controller (while this text is visible)```.

As long as the last step is followed properly, the controller will pair successfully; you need to unplug/replug the controller because BlueZ's sixaxis plugin will only display a service authorization request on USB hotplug, and only if there's a Bluetooth session running that's configured to use the ```default-agent```. The PR sets up an appropriate bluetoothctl session that will wait for the service authorization prompt, then automatically authorize and trust the controller.

The PR does more than simply change the Bluetooth module's scripting, however. It's also necessary to install a newer version of BlueZ to gain compatibility with Shanwan and Gasia clones, as well as some changes to the ```hid-sony``` kernel driver to fix Gasia pairing as well as some other important usability changes.

msdos, 2019-04-01:
I have a keyboard and a PS3 controller. I've configured the keyboard to help install the PS3 controller.

I have a Clone shanwan.

I followed https://github.com/RetroPie/RetroPie-Setup/wiki/PS3-Controller#using-the-retropie-gui-to-configure-the-ps3-controller-bluetooth-connection: installed the ps3 driver. After installing, still in the GUI dialog, I selected "Configuration/Tools", then "Install/Pair (Clone support Shanwan)".

I exited the GUI installation script. In EmulationStation, pressed "Start" to be able to select "Configure Input". Here, I plugged the PS3 controller in USB mode, waited 10s (the controller don't stop vibrating), removed the usb and pressed the bluetooth button on the controller. It worked.

The discussion in https://retropie.org.uk/forum/topic/18920/shanwan-ps3-controller-vibrating helped me to make it work.

LeeNX, 2019-08-01:
I have a ```SZMY-POWER CO.,LTD. PLAYSTATION(R)3 Controller```, which is not working with RetroPie v4.5.1 . 
  
Wonder if there might be worth testing anything else? More info in the forum - https://retropie.org.uk/forum/topic/23076/ps3-controller-shanwan-retropie-4-5-1-via-bluetooth-issue/2 . 
  
Ping me if I can test anything - @psyke83