RetroPie/RetroPie-Setup #2478: lr-Tyrquake Bug

actraiser30, 2018-09-02:
Hi, hi i think i found a bug,
Upon exiting of quake 1 by press quit and y for exit it launches Retroarch instead of return to emulation station.this is happening also to both mission packs and the episode 5.

joolswills, 2018-09-03:
you didn't follow the instructions when opening an issue

>Please start a topic on the RetroPie forum before opening an issue - https://retropie.org.uk/forum/

>This includes edit suggestions for the wiki. There are more people to help on the forum.

>Once a problem has been verified on the forum, an issue can be opened here.

in this case if verified you will probably be directed to open an issue on the libretro bugtracker.

closed 2018-09-03 by joolswills

actraiser30, 2018-09-04:
i`m not going to start anything in the forum is a labyrinth in there with the unlimited topics ,i give you the issue here and you can fix it if you can so simple.