RetroPie/RetroPie-Setup #2480: eduke32 Music not starting automatically

Caltaron, 2018-09-04:
When starting eduke32 the menu does only have sound effects but no music. Music is working in general because it it possible to kickstart it by setting music to disabled and enabled again in the game‘s settings. Problem gets worse when starting a game or switching a level: every time the music is off and needs manual activation.

Problem appears in eduke 32’s today’s latest binary (sorry for not having the version number at hand).

It works in latest source, but this has other problems like ignoring the resolution configured in the local config file and bad performance after changing the settings in-game. It also worked under RetroPie 4.3 in a version 4-5 months ago.

Configuration with problem:
Pi Model: 3B+
Power supply: 2,5 A
RetroPie: 4.4
Built from: Pre made SD image from RetroPie website, eduke32 installed (latest binary) via package manager
Problem seems to occur in all versions of Duke Nukem Files including AddOns.
USB Devices connected: Logitech mouse USB wireless dongle
Controller: PS3-Controller (wireless, Bluetooth)

actraiser30, 2018-09-13:
Hi can you try on another raspberry pi if you have one eg: raspberry pi 2

actraiser30, 2018-09-25:
Do a full os update eg apt-get-update / apt-get upgrade apt-get dist-upgrade and try again use the install from source option for eduke32

actraiser30, 2018-09-25:
Do a full os update eg apt-get-update / apt-get upgrade apt-get dist-upgrade and try again use the install from source option for eduke32
install also timidity,xorg,xinit

actraiser30, 2018-09-25:
Do a full os update eg apt-get-update / apt-get upgrade apt-get dist-upgrade and try again use the install from source option for eduke32

install also the packages under:

sudo apt-get install build-essential nasm libgl1-mesa-dev libglu1-mesa-dev libsdl1.2-dev libsdl-mixer1.2-dev libsdl2-dev libsdl2-mixer-dev flac libflac-dev libvorbis-dev libvpx-dev libgtk2.0-dev freepats timidity

joolswills, 2018-10-23:
This is not a problem with the latest binary/source.

closed 2018-10-23 by joolswills