RetroPie/RetroPie-Setup #2483: Latest Reicast does not compile on Ubuntu 16

thatjoshguy, 2018-09-13:
Due to several changes to the source, the latest release of Reicast will not compile on Ubuntu 16.04 via the "optional packages" script.

By modifying RetroPie-Setup/scriptmodules/emulators/reicast.sh like so:

from:
```
if isPlatform "x11"; then
     gitPullOrClone "$md_build" https://github.com/reicast/reicast-emulator.git
```

to:
```
if isPlatform "x11"; then
     gitPullOrClone "$md_build" https://github.com/reicast/reicast-emulator.git r7
```

RetroPie will download and compile the previous r7 release, which works fine.

cmitu, 2018-09-18:
This is an upstream problem (https://github.com/reicast/reicast-emulator/issues/1375) which should be resolved there, there is no reason to downgrade every x86 installation to the r7 version.

thatjoshguy, 2018-09-18:
Fair point and I agree, but in case they drag their feet I offer a temporary solution for those who need it.

cmitu, 2018-09-18:
They can 'drag their feet' as much as they want to, I'm sure they have their reason.

joolswills, 2018-09-18:
@thatjoshguy Please follow our procedure first though as detailed when opening a ticket (we ask people to use the forum first). I would also add that we cannot test and support old LTS releases so I am going to close this

closed 2018-09-18 by joolswills

dmschwartzii, 2018-09-30:
Issue fixed upstream