RetroPie/RetroPie-Setup #2503: Update Amiberry binary install

dudleydes, 2018-10-08:
Currently the binary install of Amiberry from the setup script installs version 2.18.2 ([25 March 2018](https://github.com/midwan/amiberry/commit/a8715535dd980db375875292e50cb943d5ac147e)).

This version breaks the joystick auto-configuration in Stretch, i.e. RetroPie 4.4, due to this [bug](https://github.com/midwan/amiberry/issues/300). A fix was implemented on [15 April](https://github.com/midwan/amiberry/commit/984f9d07f971779f25819acaecc7fe2388930229) with the release of version 2.19.

This issue comes up regularly at the forum, this [thread](https://retropie.org.uk/forum/topic/19535/amiberry-whdload-mouse-joystick-config) being the latest, where users have been able to fix by installing from source.

joolswills, 2018-10-09:
Thanks. I will rebuild the binary.

dudleydes, 2018-10-09:
It may be worth waiting to rebuild.

There is another [bug](https://github.com/midwan/amiberry/issues/353) which was fixed yesterday, with the commit being pushed to the `dev` branch.

I can look out for when the `dev` branch is merged with the `master` and let you know when this has happened.

closed 2018-10-23 by joolswills