RetroPie/RetroPie-Setup #2506: lr-mupen64plus build fails on rpi2/rpi3

joolswills, 2018-10-12:
Looks like a makefile issue, as some neon asm is not being built/linked correctly. Older sources work.

Currently bisecting.

hhromic, 2018-10-12:
I was also trying to debug this, great you are taking care.

joolswills, 2018-10-13:
It's this commit https://github.com/libretro/mupen64plus-libretro/commit/e54a7279d5610602177c3fcaa6f747a7c0547cfd

The logic as it is now, platform gets "unix" added to it for RPI, and then the rpi logic later isn't included.

I have a fix which I will open a PR for.

joolswills, 2018-10-13:
PR opened upstream

https://github.com/libretro/mupen64plus-libretro/pull/73

hhromic, 2018-10-13:
Great job @joolswills, your PR got already merged upstream and I can confirm `lr-mupen64plus` is compiling fine again from source on my RPI3b+. Cheers.

closed 2018-10-13 by joolswills