RetroPie/RetroPie-Setup #2508: Fix package install failure when libraspberrypi-dev is also included

paulwratt, 2018-10-15:
I have Devuan Jessie running on a RPi2

It has `libraspberrypi-bin` and `libraspberrypi0` installed, but not `libraspberrypi-dev`

Retropie-Setup wants to install the package `libraspberrypi-dev` but it is not in the Devuan repo (they are manually installed when the image is create, along with the kernel), so for the following messages the installations fail, even though the other packages are present and would be installed properly if `libraspberrypi-dev` was not in the package list:
```
Did not find needed package(s): libsdl2-dev libusb-1.0-0-dev libraspberrypi-dev. I am trying to install them now.
Did not find needed package(s): dh-autoreconf libibus-1.0-dev fcitx-libs-dev libraspberrypi-dev. I am trying to install them now.
```

If you placed any `libraspberrypi` package requests in a seperate command process, the above would not have failed.

Cheers

Paul

joolswills, 2018-10-15:
we neither test nor do any development on Devuan for RPI. Please follow issue tracker instructions when opening a ticket - you should start a topic first on the forum.

The packages that have that as a dependency require it for Raspbian - as it contains development libraries needed. You should report this upstream to Devuan RPI, or submit a patch to this project if the upstream package is renamed or the development libs/headers are installed via another means.

Not something I will fix. Sorry.

closed 2018-10-15 by joolswills

paulwratt, 2018-10-15:
it is not a problem to have Retropie-Setup try to install `libraspberrypi-dev`, the problem is that it does not exist on many platforms that the RPi supports. So when that package name is combined with others the whole package selection fails only because `libraspberrypi-dev` fails.

If any need for `libraspberrypi*` were seperated to a single (`apt-get install`) request, then Retropie-Setup would work on more RPi/armhf platforms, and the logs would only show that `libraspberrypi*` failed, not nay that re in the OS repo.

joolswills, 2018-10-15:
That wouldn't be the correct fix. The correct fix is to put in some code to change the dependency to whatever the equivalent is on Devuan that contains the RPI headers and link libs. If you are unable to provide a patch I recommend you start on the forum and see if there are any other uses who would get involved in contributing one, as is indicated when opening an issue here (We ask you start a topic on the forum before opening a ticket which you ignored twice).

hhromic, 2018-10-18:
Given that supporting Devuan seems to need more work here and in #2509, in other words RetroPie currently doesn't install packages correctly in Devuan, maybe is wiser to revert PR #2507 ?
Users might be misled that Devuan is fully supported when actually is broken.

joolswills, 2018-10-18:
As far as I know it works on Devuan for x86/x86_64 fine. If Devuan has renamed packages for the RPI it can be resolved by someone submitting a PR - I won't be looking at it any time soon nor have the time to support Devuan on RPI.

#2509 is not a bug and should have been a topic on the forum (The author chose to ignore the instructions when opening an issue).

I will accept a PR to correct the missing package for Devuan RPI, but don't have time to look at it myself.

joolswills, 2018-10-18:
I have disabled RetroPie-Setup for Devuan on the RPI. It will now throw an error on install. Should work fine on Devuan for x86/x86_64 (Although some modules may rely on systemd and fail to work correctly).

hhromic, 2018-10-19:
Right, #2509 was pointing out a wider scope, not just Devuan, and I also overlooked that RetroPie is run from x86/x64 machines too. I think your solution of disabling for RPI is good enough.
I also agree with the word of caution about systemd. A quick grep shows that `splashscreen`, `usbromservice`, `autostart`, `xarcade2jstick` and `bluetooth.sh` one way or another interact with systemd.
And I'm happy you decided to start dropping `jessie`, is very outdated by now and users should be encouraged to migrate, specially if they intend to keep things up-to-date. Good stuff!