RetroPie/RetroPie-Setup #2509: Option to install Retropie SDL1.2 and SDL2

paulwratt, 2018-10-15:
In Retropie-Setup I could not find the GUI options to install SDL1.2 or SDL2.

Any RPi OS should be able to install the binary packages directly. However to build from source the OS needs to have the following packages installed:
```
libraspberrypi0
libraspberrypi-bin
libraspberrypi-dev
```
As far as I know those are only available in the RaspberryPi repo, or without a RPi kernel, in the Raspbian repo. They are usually installed manually when setting up a OS filesystem image for an SD-card. So sometimes `libraspberrypi-dev` is missing (eg Devuan Jessie).

Any OS should be allowed to install the Retropie versions of SDL1.2 and SDL2 from source.

Any OS on a RPi should should be able to do the same, as long as the correct version of the `libraspberrypi-dev` package is installed. As long as the OS can use `dpkg` then `dpkg -l |grep raspberrypi-bin` will show the version needed.

Anyway, Retropie will go through the complete `core` (`base`?) installation before letting you know that `libsdl2-dev` can be found (see seperate issue with `libraspberrypi-dev` in the package list). This to me seems completely wrong, because most binaries that it builds need the SDL-dev headers anyway.

NOTE: in this example the OS is Devuan Jessie for RPi2, which had SDL 1.2.15-10 with -dev, and SDL 2.0.2 without -dev, installed before running Retropie-Setup

joolswills, 2018-10-15:
Please start a topic on the RetroPie forum before opening an issue - https://retropie.org.uk/forum/

>This includes edit suggestions for the wiki. There are more people to help on the forum.

>Once a problem has been verified on the forum, an issue can be opened here.

>Please remove this text before posting.

joolswills, 2018-10-15:
We don't officially support Devuan either - it was a 3rd party contribution to allow RetroPie-Setup to be run on this OS. If you want a change of behaviour, you will need to submit a PR.

closed 2018-10-15 by joolswills