RetroPie/RetroPie-Setup #2515: runcommand - disable menu also disables joystick for default emulator selection

hhromic, 2018-10-20:
In my setup, I have the runcommand menu disabled with `DISABLE_MENU=1` in the configuration.
When there is no default emulator for a system, runcommand anyway show the dialog for selecting a default emulator (I believe this is ok to happen!), however this dialog accepts no input from the connected joystick (can't select any option nor move). I have `DISABLE_JOYSTICK=0`.

In the code for `runcommand.sh` I found this line: <https://github.com/RetroPie/RetroPie-Setup/blob/5af30b868318b2fdb4d18b87607c7781757434f0/scriptmodules/supplementary/runcommand/runcommand.sh#L93>
Which force-disables the joystick when the menu is also disabled.
I don't know the reason for this condition in the code, however it is preventing the default emulator selection dialog to be used with a joystick. Removing the line solves the issue.
I didn't want to make a PR without raising the issue for consideration first.

steeeb, 2018-10-22:
I think disable_joystick should relate purely to the joystick to get the runcommand menu up in the first place (by pressing A).

So I think this should be fixed.

joolswills, 2018-10-22:
it just dates back to older code - probably didn't take into account the menu to select default emulator - which isn't something that comes up often. I'll probably just remove the line.

@steeeb I don't want to change the logic like that as some joystick input devices could technically interfere with navigation. At least it was a problem in the past - it may not be now.

closed 2018-10-22 by joolswills

hhromic, 2018-10-23:
Thanks for the heads-up @joolswills , good that it was just obsolete code, thanks for the fix.
In my tests this change seems harmless so far and fixes the joystick in the default emu dialog, so should be fine.

commit 60405b03 (2018-10-22): runcommand - don't disable joystick if menu is disabled - fixes #2515
 * if there is no default emulator chosen then you will not be able to navigate the default emulator menu.