RetroPie/RetroPie-Setup #2519: Add Raine Emulator to Retropie

actraiser30, 2018-10-23:
Hi is it possible to add the Raine emulator to RetroPie?

Raine is a arcade emulator which emulates some games which use M68000 or M68020 CPU's. The main focus is on emulating Taito and Jaleco games.The emulator is well written and is capable of emulating some games which MAME can't emulate.

https://github.com/zelurker/raine

Thanks

joolswills, 2018-10-23:
Please use the forum as previously directed.

closed 2018-10-23 by joolswills

actraiser30, 2018-10-23:
i cant open a discussion for this,as nothing happened with my previous posts there (extra-ports-kodi shortcut).so its better to address this to the creator here

actraiser30, 2018-10-23:
Its no use to open a discussion as nothing happened with my previous posts there (extra-ports-kodi shortcut).so its better to address this to you directly

joolswills, 2018-10-23:
That's not how it works.

actraiser30, 2018-10-23:
i understand no problem.

actraiser30, 2018-10-23:
i understand no problem.
At least if its not possible to add a shortcut to kodi after installation is finish,you can add an extra option in the kodi installation menu to add a shortcut, so whoever wants can add it or not