RetroPie/RetroPie-Setup #2542: Emulator specific retroarch.cfg keeps parameter inside file when you 'unset' it via GUI

movisman, 2018-11-26:
Hi Jools,

Not sure if intentional, but when you 'unset' a value in an emulator specific retroarch.cfg using the RetroPie-Setup GUI, it keeps the parameter in the file, but with no value, for example:

Prior to edit:
input_overlay_enable = "true"

If I edit this in the GUI, and change it back to 'unset' because I want to remove it, it keeps this in the retroarch.cfg file:
input_overlay_enable = ""

I don't believe this causes a problem, but it does mean the files could get messier than one would like.

Would it be fair to suggest, that if a parameter is changed to 'unset', it removes them completely in emulator specific retroarch.cfg files?

When you 'unset' a value in the top level all/retroarch.cfg file, it returns it to the default value, which is perfect. It is just these emulator specific ones where I think the behaviour would be better if the parameter is removed after unsetting it.

Any thoughts?

Cheers

joolswills, 2018-11-26:
This is by design. I don't want to change the way this works. Also please follow the instructions when opening an issue (you should know this already)

>Please start a topic on the RetroPie forum before opening an issue - https://retropie.org.uk/forum/

closed 2018-11-26 by joolswills

movisman, 2018-11-26:
Hi,

Yep, no worries and thanks, didn't realise it was by design. Apologies for opening on here first, I do know the protocol but at the time genuinely thought this was an issue and not one which I needed to necessarily open a thread for. Which is annoying for you, I know. I won't raise any more issues on github unless you've instructed me to via the forum - will always make a thread first in future and see what response I get there.

Cheers man.