RetroPie/RetroPie-Setup #2543: Interface Issue with CONFIGURING Controller screen

CyberManifest, 2018-11-27:
I'm getting: https://imgur.com/UJhf6wk

(No X, Y, Thumb, Trigger, Shoulder, etc. In Controller Configuring Window) (But oddly "Page Up", and "Page Down"?)

Instead of: https://i.ytimg.com/vi/Oa0JtIyhNAE/maxresdefault.jpg

(Normal expected use.)

On a fresh install of RetroPie into an existing system (Raspbian Lite on SSD using element14 Pi Desktop HAT), using this guide:

https://retropie.org.uk/docs/Manual-Installation/

Attempted Uninstall, Reinstall.

Attempted copying a known working controller config from a USB Flash Image Setup of RetroPie

Attempted Updating

Nothing Seems to fix; could you please tell me how to fix this?

joolswills, 2018-11-27:
>Please start a topic on the RetroPie forum before opening an issue - https://retropie.org.uk/forum/

>This includes edit suggestions for the wiki. There are more people to help on the forum.

>Once a problem has been verified on the forum, an issue can be opened here.

>Please remove this text before posting. (you did this but ignored the instructions above)

closed 2018-11-27 by joolswills

CyberManifest, 2018-11-28:
I couldn't find a proper section in the forum, and I couldn't post because I am not a member, and I don't wish to sign up for yet ANOTHER service. This section is labeled "Issues" so I'm reporting my "issue" here.

joolswills, 2018-11-28:
https://retropie.org.uk/forum/category/8/help-and-support is the section to post in. You will need to start there first if you want to get help.