RetroPie/RetroPie-Setup #2544: lr-mupen64plus is not building in current master

hhromic, 2018-11-27:
Hi @joolswills , since a while now `lr-mupen64plus` is not building for RPI in current master.

The reason is that the Makefile broke again in a recent PR. This is the same problem that you addressed before in PR https://github.com/libretro/mupen64plus-libretro/pull/73.

As you know, there is a long-standing PR from @psyke83 that addresses old libnames being used and he recently updated it to also fix the build again: https://github.com/libretro/mupen64plus-libretro/pull/39
However, the upstream maintainers seem to be away for the moment and the PR hasn't been merged.

I think there are three courses of action for this:
1) Locally patch the Makefile to fix the build, i.e. applying @psyke83 's PR from the scriptmodule.
2) Set the build to a known-to-be-good commit (similar to what you did in `lr-parallel-n64` recently). In this case the commit would be the one before the PR that broke the build: `bb64262862`.
3) Try to poke again the upstream maintainers to merge the PR from @psyke83 (ideal solution). Perhaps you have more influence over this than me :)

joolswills, 2018-11-27:
Yeah I think we have no choice but to include the patch. Not sure upstream are listening right now.

Shame they don't review PRs :/

hhromic, 2018-11-27:
Indeed a shame. To save you some time I just quickly added the extra patch and I'm testing the build now. After confirming its working fine I'll send a PR.

PR #2546 (2018-11-27): lr-mupen64plus - add patch from @psyke83 to fix broken build and refresh existing patch
Tested and building fine now. Fixes #2544
commit b5f19e81 (2018-11-27): lr-mupen64plus - add patch from @psyke83 to fix broken build and refresh existing patch

fixes #2544
closed 2018-11-27 by joolswills