RetroPie/RetroPie-Setup #2556: Support for next Amiberry version

midwan, 2018-12-06:
Hi guys,

I'm getting ready to publish a next "stable" release of Amiberry, which will come with binaries for all supported platforms. My aim with this Issue is to ensure it all goes smoothly for RetroPie users also.

There are a few things I'd like to address, to get the best possible experience:
- I noticed that you never build the RPI3 version of Amiberry. There's only a check if the platform = rpi1 otherwise it defaults to rpi2. However, the RPI3 version has different compiler flags which have been shown to improve performance further, compared to the RPI2 build. I would therefore recommend that you check if the platform = rpi2 or platform = rpi3, and target those accordingly.

- I'm considering changing the default platform to `rpi3-sdl2-dispmanx` instead of `rpi3` as it is currently. As most platforms are running the SDL2 version today, I'd like to harmonize that in the Makefile also. In fact it would be great if we could remove the SDL1 version sometime in the future, as it offers no advantages over the SDL2 one, and it's actually a little slower also. This change shouldn't affect your environment, as from what I've seen in the scripts you explicitly specify the platform during building always. So if that's the case, it's just FYI.

- The binaries I'll provide will be further optimized using a profiler, so they will show improved performance when compared to a new compile from source. They will also be compressed using `upx`, so their size will be smaller. Again, this is just FYI in case you get questions about that in the future. :-)

- Do you have a preference regarding the format the binaries are delivered? E.g. compression method (zip/tar.gz), filenames? Is the current approach good enough for you?

Finally, a little extra information just to clarify: there are currently 3 different offerings of Amiberry for RPI3:

1. SDL1 (using Dispmanx)
2. SDL2 (using Dispmanx)
3. SDL2 (no Dispmanx, using pure SDL2 with whatever backend it was configured with)

The best performing one of these, is currently **2 - SDL2 with Dispmanx**, and specifically with SDL2 configured to use the **KMSDRM** back-end. 

In order to get that, you will need to use the **fake KMS** driver on the RPI, and **NOT the full KMS one** - that will disable **DispmanX** completely. If you're using the Legacy driver, I believe you cannot use KMSDRM with SDL2 (I could be mistaken, it's been a while since I tested this config). 

Also, if you have SDL2 configured to use the `rpi` back-end, you will run into problems:
Amiberry uses Dispmanx directly for the emulation screen, for maximum performance. But SDL2 will also use Dispmanx if configured with the `rpi` back-end, so Amiberry's GUI will open up using that. Once SDL2 locks Dispmanx however, the emulation screen will fail to open after the GUI, so you'll get a black screen and perhaps even an unstable system.

hhromic, 2018-12-07:
> I noticed that you never build the RPI3 version of Amiberry. There's only a check if the platform = rpi1 otherwise it defaults to rpi2. However, the RPI3 version has different compiler flags which [...]

This is true for everything in RetroPie not just amiberry, and I second your recommendation that maybe it is time to start compiling natively for Cortex-A53 processors (RPI3 family). The compiler flags for rpi3 platform are there but not used for shipped binaries for simplicity and more testing requested:
```
# note the rpi3 currently uses the rpi2 binaries - for ease of maintenance - rebuilding from source
# could improve performance with the compiler options below but needs further testing
function platform_rpi3() {
    __default_cflags="-O2 -march=armv8-a+crc -mtune=cortex-a53 -mfpu=neon-fp-armv8 -mfloat-abi=hard -ftree-vectorize -funsafe-math-optimizations"
    __default_asflags=""
    __default_makeflags="-j2"
    __platform_flags="arm armv8 neon rpi gles"
}
```
However it seems that you have a good argument that indeed compiling for rpi3 brings better performance. I personally compile retroarch and lr cores from source to take advantage of the compiler flags for my RPI3B+, and the resulting binaries are very stable. Therefore I can advocate stability, but I have never done a proper benchmarking between rpi2 and rpi3 binaries like you seemed to have done.

Looking forward to the consideration of shipping cortex-a53 binaries!

CookiePLMonster, 2018-12-08:
Wow, that is very surprising... **not** providing binaries tailored for rpi3 kind of defeats the main advantage of using unified HW system, that is tailoring for a very specific CPU. If binaries for rpi and rpi2 are provided, it feels like there is absolutely no reason not to provide them also for rpi3.

Thanks for the heads up about this, I'll make sure I keep a close eye on this next time I decide to update from binary vs from source.

joolswills, 2018-12-08:
I've not seen any noticeable performance increases from testing with rpi3 vs rpi2 flags on a rpi2. eg benchmarking with tyr-quake. Please can you provide some reproduceable benchmarks.

>I noticed that you never build the RPI3 version of Amiberry. There's only a check if the platform = rpi1 otherwise it defaults to rpi2.

where did you see this ?

We ship our own binaries generally, and I don't want to change that. It takes time to maintain all the binaries, but I don't want to start installing binaries maintained by 3rd parties unless we have to (eg - closed source projects).

>Wow, that is very surprising... not providing binaries tailored for rpi3 kind of defeats the main advantage of using unified HW system, that is tailoring for a very specific CPU. 

Not really - the rpi3 runs in 32bit mode for RetroPie so it's not like the difference between armv6 -> armv7+neon

CookiePLMonster, 2018-12-08:
> Not really - the rpi3 runs in 32bit mode for RetroPie so it's not like the difference between armv6 -> armv7+neon

No new instructions for 32-bit mode? No new assumptions about efficiency of specific instructions (eg. like signed/unsigned operations in early ARM boards)?

joolswills, 2018-12-08:
Yes. Since I maintain them and without evidence of improvement there is no point rebuilding everything. If someone would show that certain binaries are noticeably faster, I may consider building a few rpi2 specific binaries. That's my final decision.

CookiePLMonster, 2018-12-08:
But if it's automated then it costs no additional time, other than building time?

Then again, like you said it largely depends on the differences between CPUs. Now if x32 (arm64 instruction set with 32-bit memory pointers) was possible then it'd make a noticeable difference everywhere. Sadly, the concept of x32 seems largely neglected around the entire Linux ecosystem.

midwan, 2018-12-08:
> I've not seen any noticeable performance increases from testing with rpi3 vs rpi2 flags on a rpi2. eg benchmarking with tyr-quake. Please can you provide some reproduceable benchmarks.
> 

I benchmarked the binaries a few years ago, when I first started Amiberry, and it showed some performance improvement with the RPI3 build vs the RPI2 build. I don't have those numbers at the moment, but I'll happily run a few new ones with the latest version and get back to you with more concrete results.

> > I noticed that you never build the RPI3 version of Amiberry. There's only a check if the platform = rpi1 otherwise it defaults to rpi2.
> 
> where did you see this ? from source we do - only for binaries do we provide rpi1 and rpi2 bins.
> 

I noticed that when upgrading the binary from within RetroPie-Setup. Sorry I wasn't clear enough in my description above. Building from source does indeed get the right version for the platform.

> We ship our own binaries generally, and I don't want to change that. It takes time to maintain all the binaries, but I don't want to start installing binaries maintained by 3rd parties unless we have to (eg - closed source projects).
> 

I wasn't aware that you ship your own binaries, I thought you were getting it from my repo's releases instead. Like I mentioned originally, I go through some extra effort to further optimize the binaries I provide, by running them through a profiler twice, for maximum performance. I'm ok if you want to maintain your own binaries of course, this is your project and you decide how to best handle things - I'm just offering options so you can make the best educated decision. :)

I'll get back to you with benchmark results a little later.

CookiePLMonster, 2018-12-08:
> I benchmarked the binaries a few years ago, when I first started Amiberry, and it showed some performance improvement with the RPI3 build vs the RPI2 build.

Do you know what exactly are the changes made by that switch? If we knew exactly what the switch changes in code generation, a more theoretical approach could be possible.

midwan, 2018-12-08:
@CookiePLMonster We could examine the Assembly files the compiler generates between the two different targets, but that would take quite some time to go through, and I'm not sure we would reach a conclusion easily that way. I am a bit pragmatic in such things, so I prefer to run multiple real-life tests and examine the performance that way.

midwan, 2018-12-08:
@joolswills 
Here are some benchmark results, as promised:

**Setup:** since synthetic benchmarks tend to be unreliable, I run a real-life scenario that took considerable time to complete, and repeated the experiment 4 times on each platform and with each binary. The test was rendering a specific Lightwave scene I have prepared for benchmarking purposes.

**Platforms:** I tested the same scene on the Raspberry Pi 3, and Raspberry Pi 3B+

**Binaries:** The binaries tested were the `rpi2-sdl1`, `rpi3-sdl1`, `rpi2-sdl2-dispmanx` and `rpi3-sdl2-dispmanx`. The version tested was the  current one from the `master` branch.

**Configuration:** Since the test focuses on raw (emulated) CPU power, the emulator was configured accordingly by setting the CPU mode to Turbo, JIT and JIT FPU, Frameskip, disabling Sound and Collision emulation. The identical configuration was of course used across all platforms.

**Precautions:** To avoid any issues from overheating (which would cause an underclock on the board's CPU), there was a 5 minute pause before each test to allow the temperatures to drop back to normal.

**Test measurements:** The number of seconds it took to complete the scene, is reported by Lightwave itself. Less is better of course. To ensure that we get a reliable measurement (e.g. the numbers fluctuate a bit due to the JIT engine and other factors), the tests were repeated at least 4 times for each binary and platform. The best result was kept, while I kept track of the rest of the numbers also.

**Results:**

- RPI3, using the rpi2 binary: Best result was **329 seconds** after 4 runs.
- RPI3, using the rpi3 binary: Best result was **325 seconds** after 4 runs.
- RPI3B+, using the rpi2 binary: Best result was **307 seconds** after 4 runs.
- RPI3B+, using the rpi3 binary: Best result was **300 seconds** after 4 runs.

So granted, there's not a huge difference but there certainly is a measurable one. As I said, the numbers did fluctuate a bit (+/- 3 seconds) per run, so these are the best results each platform and binary combination produced.

CookiePLMonster, 2018-12-09:
Interesting results, since I'd risk saying the difference is big enough to say that "_compiling for rpi3 does not make it slower_" but not big enough to say that "_compiling for rpi3 makes it faster_".

joolswills, 2018-12-09:
Thanks for the in depth analysis. Interesting. Measurable for sure, although not enough to persuade me to maintain RPI3 bins. :-) but appreciated.

It's worth mentioning that raspbian also only ship with armv6 and armv7 bins.

I may revisit in the future and do have plans to improve our updating mechanism so packages installed from source will be updated from source (our current system is rather basic).

I'd be interested in results with profiling. According to a user in the forum's they had a few FPS gains with snes9x with profiling.

hhromic, 2018-12-09:
@midwan , indeed interesting results, I have been looking for a while for comprehensive benchmarking of this precise scenario. Just for the sake of completeness, can you share the exact compiler flags you used for generating the binaries for rpi2 and rpi3? thanks.

> I may revisit in the future and do have plans to improve our updating mechanism so packages installed from source will be updated from source (our current system is rather basic).

I have been thinking myself a lot about this as well, not only to provide source->source, binary->binary update paths, but also (more importantly for now) to provide a basic versioning system so unmodified upstream doesn't need to be recompiled everyt time.
Do you plan to open a forum thread to propose and discuss approaches? or an issue here? I would be very interested on participating on the design/implementation of this.

> I'd be interested in results with profiling. According to a user in the forum's they had a few FPS gains with snes9x with profiling.

Profiling indeed is the future for possibly better binaries generation, however it is not easy to automate at the moment. It would have to be implemented on a third-party provided basis. Also, apparently profiling data is not portable either (yet?) so every time the source changes, new profiling data must be generated and this definitively is time-consuming.

joolswills, 2018-12-09:
>Do you plan to open a forum thread to propose and discuss approaches? or an issue here? I would be very interested on participating on the design/implementation of this.

No :-)

I have a basic plan in my head and some code I started. I don't really want to develop this by committee. I do RetroPie for fun, and like to just get on with the code sometimes.

midwan, 2018-12-09:
Regarding profiled binaries, I run the same benchmarks today using the RPI3 binary after passing it through a profiler (with the specific Lightwave render scene). I tested the same binary on a RPI3 and a RPI3 B+, and again followed the same steps (5 minute breaks, 4 runs, keeping the best results):

- RPI3: Best result was **315 seconds**, down 10 seconds from the previous best of **325**
- RPI3 B+: Best result was **294 seconds**, down 6 seconds from the previous best of **300**

CookiePLMonster, 2018-12-09:
Using PGO for emulators could be very risky though, as then you'd end up optimizing for a subset of games and programs - potentially harming the other software using different execution patterns.

Just my two cents on PGO.

hhromic, 2018-12-09:
> No :-)
> 
> I have a basic plan in my head and some code I started. I don't really want to develop this by committee. I do RetroPie for fun, and like to just get on with the code sometimes.

No problem, I also do it for fun on my side :) Take your time and let's see things when they start to see light.

grant2258, 2018-12-18:
if thats the case why dont you just grab the bins from libretto buildbot for the lr-cores and users can compile an optimized one if they choose too. 

https://buildbot.libretro.com/nightly/linux/

https://buildbot.libretro.com/nightly/linux/armhf/  
https://buildbot.libretro.com/nightly/linux/armv7-neon-hf/

midwan, 2019-01-26:
Just FYI, the next stable version of Amiberry is now out: https://github.com/midwan/amiberry/releases
Binaries are also included for all RPI platforms.

midwan, 2019-01-26:
@joolswills 
Ah yes, then it's probably up to date already. The version should read v2.24 in the About Panel, in the emulator GUI. It was published on the repo a bit more than a week ago.

I know about the issue, I think it was about `std:ceil()` missing or similar? But since I thought we've moved to Jessie now for all supported platforms, it shouldn't matter much.

joolswills, 2019-01-26:
yeah - that was the issue. Yeah not important if you don't want to support gcc 4.x

joolswills, 2019-02-07:
I thought I had uploaded the latest bins but it was still saying beta (maybe I didn't upload after building) so I just updated now.

Can this ticket be closed? Can leave it open for future discussion if you prefer,.or open a new ticket later.

midwan, 2019-02-07:
@joolswills 
Thanks!
Let's close this one, and we can re-open a new one in the future - it would be easier to read through if we keep it short and to the point for each issue I think :)

The next item I'm looking into is dropping SDL1 and using SDL2, but in a way that would work for RetroPie out-of-the-box (with the SDL2 implementation you're using, which is compiled with the `rpi` back-end). I've already done some tests and it looks promising, but we'll take it on a separate issue when the time comes.

closed 2019-02-07 by midwan