RetroPie/RetroPie-Setup #2575: Internet Archive Support

ahundt, 2019-01-10:
The [Internet Archive](archive.org) provides access to a legal ROM collection as part of its mission for historical data preservation. 

There is a small script set up for loading these with retropie:
https://github.com/skullskullskull/Archive

Perhaps there is a way these resources could be directly integrated with RetroPie?

cmitu, 2019-01-10:
The IA has a DMCA exception to host those, but not to distribute them - see https://www.reddit.com/r/emulation/comments/4kukq1.
Such script would just make RetroPie a ROM distribution software, thus enabling piracy. See also 
https://retropie.org.uk/forum/topic/9138/roms-to-be-downloaded-on-the-retropi-os.

joolswills, 2019-01-10:
Also - please don't ignore the instructions when opening a ticket. Use the forum.

closed 2019-01-10 by joolswills

ahundt, 2019-01-10:
I have some familiarity with copyright law through personal curiosity, but I'm not a lawyer. It seems those linked discussions don't appear to be by lawyers, or part of an organization that has a lawyer. 

> The IA has a DMCA exception to host those, but not to distribute them - see
> https://www.reddit.com/r/emulation/comments/4kukq1.
> Such script would just make RetroPie a ROM distribution software, [...]

Isn't the rom downloaded by the web browser when you visit the hosting site, and run in a javascript emulator? To me, that doesn't seem very different from visiting the hosting site with a bash script executing C code, then running the file in retropie's choice of language. Does that make a difference?

The IA themselves say their [in-browser tools save you the trouble of finding your own emulator](https://archive.org/details/historicalsoftware&tab=about), perhaps that means you're able to do so for historical software hosted by a library?

(genuinely curious)

I did some more searching and [here is some news](https://www.eff.org/deeplinks/2018/11/expanded-dmca-exemption-video-game-preservation-grants-small-victory-amidst), but nothing I've found is very clear on this specific topic.

https://archive.org/details/softwarelibrary%26tab=collection&tab=about

It is entirely possible that individuals accessing ROMS from a public library (the IA) would meet the requirements of the library exemption. If you in turn distribute them yourself that could be problematic. Nonetheless, I understand the decision to be careful in your choice of what to implement, particularly if you're not employed by a library. Perhaps the only more definitive possibility would be to ask a lawyer specializing in this area, such as at the EFF or the IA itself about where the line is.

Other interesting links:

https://archive.org/about/dmca.php
https://law.stackexchange.com/questions/22078/what-are-the-implications-of-archive-orgs-dmca-exception

Sorry if this is unhelpful, I'm sure constant clarification is needed on these projects. Wouldn't it be great if it was all much simpler? One can only wish.

> Also - please don't ignore the instructions when opening a ticket. Use the forum.

Sorry about that. Would https://retropie.org.uk/forum/category/10/ideas-and-development be the appropriate spot?

Thanks for your replies.

joolswills, 2019-01-10:
We will never include anything like this even if you think it's legal.