RetroPie/RetroPie-Setup #2576: update mame2003 binary

markwkidd, 2019-01-10:
The current binary build of mame2003 has some significant bugs that have been fixed.

For example: https://retropie.org.uk/forum/topic/2612/mortal-kombat-2-3-umk3-audio/

The current binary is also prior to the vector rendering fixes.

Is it possible to update the binary to avoid more users reporting bugs that have already been fixed?

joolswills, 2019-01-10:
Use the forum.

closed 2019-01-10 by joolswills

markwkidd, 2019-01-10:
That's fine. There have been many posts already in the forum about this, so
I'm sure others with chime in.

joolswills, 2019-01-10:
Then you can tag me in a post. I have already seen the thread though - I do not need issues opened with requests to update the binaries. I keep an eye on things and update as I see fit. If you think a change is important you can tag me on the forum and I may consider it.

markwkidd, 2019-01-10:
That seems totally fair. I know the culture and workflows better over on
the libretro/RetroArch side of things -- I aspire to also doing things "the
right way" in the RetroPie domain