RetroPie/RetroPie-Setup #2588: RetroPie v4.4 Cannot Control USB Sound Volume After Update/Upgrade

tinkerBOY-git, 2019-01-28:
I want to post my issue at RetroPie forum but I couldn't. The site wont let me post. Must be an issue with AKISMET plugin.

I'm using a [Game Boy Zero Controller with builtin C-Media USB audio](https://www.tinkerboy.xyz/product/tinkerboy-controller-v3-0-game-boy-zero-aio-pcb-controller/) on a Raspberry Pi Zero, 3 B, 3 B+, and 3 A+. Sound Settings under emulationstation menu works perfect with version 4.3 and stock 4.4 and I can control the sound volume level just fine. But after updating/upgrading RetroPie v4.4 I can no longer control the volume level. I can however move the slider under Sound Settings but the actual volume on the speaker is always the same at its loudest no matter what I do.

Here's how I updated the stock RetroPie v4.4:
```
sudo apt-get update
sudo apt-get upgrade
sudo apt-get dist-upgrade
```
I also tried updating RetroPie via the RetroPie-Setup but still the same issue. I have used 3 different sdcard installation but the issue is the same.

hhromic, 2019-01-28:
Try updating EmulationStation using the developer channel `emulationstation-dev` package in the experimental section. There is a new option coming that allows you to choose the Audio Card/Device to use in EmulationStation. This is particularly necessary when using USB audio devices.

This feature is not yet in the stable branch but will be soon. In case you don't want to install the dev version, you have to just wait.

More Info: https://github.com/RetroPie/EmulationStation/pull/460

tinkerBOY-git, 2019-01-28:
I just tested the `emulationstation-dev` package but still the same issue no matter what settings i select on the "Sound Settings" the volume level on my speaker just won't change.

hhromic, 2019-01-28:
Are you sure you can control the volume of your USB device for example using `amixer`?
You may need to provide a custom control device name to EmulationStation, check the PR I referenced in my previous post, in particular about customising the device name in `es_settings.cfg`.

hhromic, 2019-02-04:
@tinkerBOY-git  did you solve your problem? can this issue be closed? thanks

tinkerBOY-git, 2019-02-04:
No that didn't solve my issue.

Gemba, 2019-02-17:
Can confirm, that you can configure a USB audio device in ES. Just tested with 8878cb0, on a RetropieSetup 4.4 (cda9ffcb)
USB device configured according to these [hints](https://github.com/RetroPie/RetroPie-Setup/wiki/Sound-Issues#usb-audio)
Had to manually enter the device name from `amixer -c 0`, in my case 'Headphone' into `/opt/retropie/configs/all/emulationstation/es_settings.cfg` as `AudioDevice`.
Then the volume of the USB audio can be adjusted. 

_However_ it does not survive reboot.
BTW: I noticed that SystemVolume is not persisted in the `es_settings.cfg`, is there a specific reason for this?

hhromic, 2019-02-19:
> _However_ the volume setting does not survive reboot, the audio device setting does..
> BTW: I noticed that SystemVolume is not persisted in the `es_settings.cfg`, is there a specific reason for this?

@Gemba , thanks for testing!  Can you migrate this issue to EmulationStation's repo instead? :)

Gemba, 2019-02-20:
@hhromic and @tinkerBOY-git , can provide some good news.

I dug deeper and found that a `systemd` issue was responsible for not storing the alsa USB audio values. Something around `/lib/systemd/system/alsa-restore.service` and `sound.target`. Did not pinpoint the cause exactly. In general it makes sense that ES is not storing the volume, since it is a OS responsibility (in this case for systemd/ALSA).

After updating systemd / raspbian (`aptitude update && aptitude -y safe-upgrade`) and now with this `systemd`:
```
$ apt-cache policy systemd
systemd:
  Installiert:           232-25+deb9u9
  Installationskandidat: 232-25+deb9u9
  Versionstabelle:
 *** 232-25+deb9u9 500
        500 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages
        100 /var/lib/dpkg/status
```
the volume of USB audio volume setting entered in ES gets persisted between reboots. :beers: 

@hhromic  Still need to reference this in ES repo for the record?

hhromic, 2019-02-20:
@Gemba, nice finding!  Thanks for researching this.

Indeed as you say this seems to be a system-specific issue and nothing to do with ES. In this case then I think it is not necessary to migrate the issue to ES as it's not related. 
Thanks again for confirming that setting audio devices in `es_settings.cfg` is working as intended and that persistence of USB devices volume requires updating the system.

@tinkerBOY-git if you still can't solve your issue, maybe you should give the forum another try to further debug what is going on with your setup.

@joolswills perhaps this issue can be closed now.

closed 2019-03-25 by joolswills