RetroPie/RetroPie-Setup #2616: 02/2019 PSX Horror Games Not Loading TRIED EVERYTHING

MikerTone, 2019-02-15:
Please start a topic on the RetroPie forum before opening an issue - https://retropie.org.uk/forum/

This includes edit suggestions for the wiki. There are more people to help on the forum.

Once a problem has been verified on the forum, an issue can be opened here.

Please remove this text before posting.

MikerTone, 2019-02-15:
Similar to so many other posts from around GitHub and the Internet in general, I've had problems getting some of my PSX games to actually play on my Raspberry Pi 3B. I have tried almost every piece of advice and tutorial and troubleshooting tip, and none of them have worked. Right now, the Spyro game I have on the Pi works fine, but the horror game Kuon won't play. In fact, none of the PSX horror games I've uploaded will work (Kuon and Fatal Frame I, II, and III - and I made sure to get the USA versions). I'll select the game from the menu, the screen fades to black, the gray box pops up saying it's loading, then it fades back to the menu. I've attempted to fix this with various long combinations of methods 3 or 4 times now, and I've started from scratch each attempt. I think I've spent just over a month on trying to get this to work.

By the end of my most recent attempt (today, February 14, 2019) I had tried:

- Making sure all the files were in the appropriate folders
- Updating RetroPie, Retroarch, Raspbian, lr-pcsx-rearmed
- Clean install of the most current versions of Raspbian and RetroPie
- Changing the cases for the BIOS files (trying both all upper and all lowercase, for 1001 and 5501)
- Making sure the BIOS files were for the same region as the game version downloaded (USA)
- Using an .ISO file (and changing the cases, both upper and lowercase) - all the horror games started as .ISOs
- Using a .BIN/.CUE file made from the .ISO (and changing the cases, both upper and lowercase)
- Double-checking the .CUE file to make sure it references the .BIN file's name correctly
- Changing file names to take out any 'special characters' (parentheses, apostrophes, etc.)
- Changing that one spot in the .CUE file from MODE1/2048 to MODE2/2352
- Installing lr-ppsspp and using a .PBP file made from the .ISO file (and changing the cases, both upper and lowercase)
- Enabling/disabling SSH
- Downloading files from a different source (and repeating many of the steps above)

The only other thing I haven't this time, but have at others, is:

- Changing file permissions to 755

Downloading the .BIN/.CUE version instead of the .ISO (I have been unable to find .BIN/.CUE versions for download despite a ton of searching - that's why I 'made my own')

Other games from other systems have also worked fine, it just seems to be my PSX horror games that won't work.

I am completely dumbfounded. I've tried to pull all the information I could find from various corners of the Internet together to figure out what could be going wrong, without any success. Is there another step or method I'm missing? Is this a current problem that doesn't have a solution yet? Or am I just lucky enough to have found 4 games that RetroPie won't play?

I'm also 100% new to pretty much all of this. I've definitely learned a lot over the past month or so, but there's probably a lot of common knowledge about Raspberry Pi / RetroPie that I don't currently have. I've tried to give the most accurate information possible above. I'm hoping someone can help me - all I want is to fight ghosts in Japanese horror games!

joolswills, 2019-02-15:
You need to use our forum as directed. Thanks.

closed 2019-02-15 by joolswills