RetroPie/RetroPie-Setup #2620: running MAME roms over CIFS requires mount options

Spongman, 2019-02-17:
It took me a while to track this down, but mame4all does `stat(2)` calls on the rom .zip files to check for their existence and/or distinguish between files and directories.

unfortunately, when you load these roms from a CIFS-mounted windows file share these stat calls return errno 75 (incorrect size). and mame4all reports "ERROR: required files are missing" and lists all the .zip file contents as not found (even though they _are_ in the location that it's looking at).

in order to fix this these mount options need to be added: `nounix,noserverino`

eg. in `/etc/fstab`:
```
//WindowsBox/roms  /home/osmc/RetroPie/roms  cifs  nounix,noserverino,ro,nobrl,_netdev,user=user,password=password 0 0
```

i would recommend updating the docs to reflect this: https://github.com/RetroPie/RetroPie-Setup/wiki/Running-ROMs-from-a-Network-Share

cmitu, 2019-02-18:
The `fstab` method (option 2 in the wiki page) already contains those options. You can add them to the wiki page for the other steps also.

cmitu, 2019-02-24:
@Spongman Thank you for adding the info to the Wiki. 
@joolswills I think this can be closed.

closed 2019-02-24 by Spongman