RetroPie/RetroPie-Setup #2626: Amiberry v2.25: support for new autoloader (.cue and .lha files)

midwan, 2019-02-20:
Hi guys,
Amiberry v2.25 is ready to release, and brings one nice new feature that would further help launching games from RetroPie.

**What's new:**
We've added a new command-line option for the autobooter, which detects both WHDLoad (`.lha`) and CD32/CDTV (`.cue`) games automatically, matches them to our internal XML database and picks the best emulator settings for them from there.
The previous option (`-autowhdload=<path-to-file>`) will remain for backwards compatibility.

The new option is `-autoload=<path-to-file>` where `<path-to-file>` is the full path to either an `.lha` file or an `.cue` cd image file. The emulator will detect what was passed and handle it accordingly (if it's a valid filename of course, otherwise it will abort).

Furthermore, the autobooter's startup performance has been vastly improved. It will now start the game within just a few seconds. Please make sure you update the `whdboot` directory contents when updating from source or binaries.

Finally, we've added a button in the emulator's GUI (under the Paths panel) to download the latest XML game database, which includes the list of known games, their best settings etc.

**What needs to be done:**
- Recognize `.cue` files and pass them to the emulator using this command-line option.
- Ideally, change `.lha` detection to also use the same command-line option, for consistency. The old one will still work, but it would be simpler in your script also to only use one call I believe.

HoraceAndTheSpider, 2019-02-20:
I should add that there are some extra files that are needed for full WHD booter functionality. Although these are included in `boot-data.zip` they are now used external of that to speed the loading process, with the former file now used as a 'fallback'
 

- ...\whdboot\WHDLoad
- ...\whdboot\save-data\Kickstarts\kick33180.A500.RTB
- ...\whdboot\save-data\Kickstarts\kick33192.A500.RTB
- ...\whdboot\save-data\Kickstarts\kick34005.A500.RTB
- ...\whdboot\save-data\Kickstarts\kick40063.A600.RTB
- ...\whdboot\save-data\Kickstarts\kick40068.A1200.RTB
- ...\whdboot\save-data\Kickstarts\kick40068.A4000.RTB

joolswills, 2019-02-22:
Thanks for the info. I will update the binary first, and switch over to the autoload later.

HoraceAndTheSpider, 2019-02-24:
Hi @joolswills 

Just thought i'd try an install with a fresh RetroPie - all looks OK with the above (the RTB files all present) but the main WHDLoad executable (\whdboot\WHDload) seems not to be installed with the install-from-source option?

it's causing an Amiga DOS error that WHDLoad is not present:

```
	WHDLoad: Unknown Command
	WHDLoad failed return code 10
```

Once added, all seems to run normally.

Cheers

joolswills, 2019-02-24:
Thanks. I'll check when I'm next free.

cesarpuig, 2019-02-27:
forgive, but in my case I think I've fixed it by restoring permissions

sudo chown -R pi:pi /home/pi
sudo chown -R pi:pi /opt/retropie/configs

2play, 2019-03-22:
hi Jim,
When updating from source the amiberry.sh does not have the above line code you mention.

it has
`elif [[ "$rom" == *.uae ]]; then
    "$emulator" -config="$rom" -G
elif [[ "$rom" == *.lha ]]; then
    "$emulator" -autowhdload="$rom" -G`

Ive manually added and cd32 boots to the game!
`elif [[ "$rom" == *.cue ]]; then
    "$emulator" -autoload="$rom" -G`

Tx for the info above. Will this be integrated or missed?

midwan, 2019-03-22:
@2play 
I'm afraid we don't have control over the `amiberry.sh` file, that's part of the RetroPie distribution. :)

I hope that it will be updated, not sure if @joolswills had a chance to look into it yet?

2play, 2019-03-22:
Noted tx @midwan  
I m using the latest from source on pi3b and all boot great wiht correct set and cue with above works like lha.

joolswills, 2019-03-22:
not yet - sorry been very busy. I will close this ticket when it's sorted.

HoraceAndTheSpider, 2019-03-26:
Hi @joolswills

Does this resolve the issue with ..\whdboot\WHDLoad not being updated/correct permissions (and therefore invisible to Amiberry) when updating from source also?

Thanks

joolswills, 2019-03-26:
No - I wasn't able to reproduce that before and forgot to check again. Will look into it.

Please can you give me some more information how to reproduce the issue ?

joolswills, 2019-03-26:
I re-read this issue and I think I just need to add one file to be installed. Will do that now.

joolswills, 2019-03-26:
Should be ok now. Please can you check/confirm?

HoraceAndTheSpider, 2019-03-28:
HI @joolswills  thanks for looking into this - i actually added a fallback (requiring a different dependency) in Amiberry to cover this file not being present, but i will try to test it by deliberatly removing the fallback.

Thankfully, if the file is now updated, we can have the WHDload binary updated correctly from within Amiberry itself, so i'll be glad to check this when i get the chance.

joolswills, 2019-04-03:
Off topic but mentioning here while it's on my mind. 

I use amiberry all the time (thanks!), although not with whdload. Mostly just launching adfs. May get around to a more efficient set up at some point.

Anyway - when saving a custom config for a game, it would be useful if it could have an option to save it in the same folder as the launched adf. That way it would be automatically used if launching that game (we check for a .uae file in our launch script).

Or even just ability to choose config folder in gui like a "save as". I made a config for a hd dir. Had to move it to ROM folder so I could use it to launch for example.

Not a biggy. Just an idea.

joolswills, 2019-04-03:
Which also makes me think we could check for a .uae config that matches base name of disk in the current config location. Which would also solve it. I think I'll do that. "Save as" could still be useful though.

HoraceAndTheSpider, 2019-04-03:
@joolswills I will be looking to add ADF support to the autoload function in the future. It will do as you say, check for a matching .uae file but in the `conf` Folder.

I would also like it to insert  multiple disks to drives 2-4 where available. 

It will do the basic checks in the file name for a1200 / AGA tags etc to set a higher setup if needed, but given the large number of ADFs out there I doubt I will do a full compatibility XML like WHDLoad (maybe, but I won’t be able to maintain that so easily!)

joolswills, 2019-04-03:
Thanks. Right now RetroPie supports booting from zips/adfs and also passes additional disks to other drives (via trying to look for the names of disks 2/3/4 etc). If you implement that natively please let me know,as  we will need to adjust our launch code to remove the way we do it currently and use your functionality instead.

Thanks!

joolswills, 2019-04-03:
For reference: https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/emulators/uae4arm/uae4arm.sh#L18-L55

joolswills, 2019-04-03:
Forgot to add - We also choose between an a500 config or a1200 config depending on the filename of the adf. If it contains AGA then an a1200 config will be used.

HoraceAndTheSpider, 2019-04-03:
Yep, perfect. The idea would be to make the RetroPie launch side as simple as possible.

The only one I’m unsure of is ZIPS which I am unlikely to add unless I can read the data out of it in-emulator (which I am sure is possible but might take some time)

mrfomt, 2019-04-14:
When I start a que file to autoload a CD32 game, the game starts and everything works, but the display is cut off on both sides, left and right, so cant see the whole game. If I go in to the settings and change the display to set it correct, then when I save the config file I loose my controller in the game, cant exit amiberry or enter config anymore. The other games still works with controller, but if I just save then the game they looses the controller.

br4nnigan, 2019-06-14:
Has Amiberry been removed? It's not in my optional packages. I tried updating setup script (from 4.0.12 or so to 4.4.14) but it's still not there. Installing UAE4Arm fails with an error.

mrfomt, 2019-06-14:
@br4nnigan It has moved to main packages

commit e76f1b6c (2019-03-25): amiberry - added support for .cue files via new -autoload parameter (and also use for lha) - fixes #2626
closed 2019-06-15 by joolswills