RetroPie/RetroPie-Setup #2662: Outdated ROM extension information

HNJAMeindersma, 2019-03-20:
It look like there is some outdated or contradicting information regarding ROM extensions on the official RetroPie-Setup Wiki.

**Commodore 64 / VIC 20**
https://github.com/RetroPie/RetroPie-Setup/wiki/Commodore-64-VIC-20-PET - The emu table contains _.crt .d64 .g64 .t64 .tap .x64 **.zip** .prg_, however the ROM chapter contains _.crt .d64 .g64 .t64 .tap .x64 .prg_
https://github.com/RetroPie/RetroPie-Setup/wiki/Supported-Systems -  The supported systems page contains _.crt .d64 .g64 .t64 .tap .x64 .prg_

**Nintendo DS**
https://github.com/RetroPie/RetroPie-Setup/wiki/Nintendo-DS - The emu table and ROM chapter contains _.nds and **.zip**_
https://github.com/RetroPie/RetroPie-Setup/wiki/Supported-Systems - The supported systems page contains _**.bin** and .nds_

Some forums give me the impression that _**.7z**_ should also be supported for the DS. I have, however, not been able to test this (yet).
https://www.reddit.com/r/RetroPie/comments/997v34/unable_to_launch_7z_roms/
https://retropie.org.uk/forum/topic/19/emulation-station-not-reading-7z-snes-roms/8

dankcushions, 2019-03-20:
personally i think we could remove the extentions column in https://github.com/RetroPie/RetroPie-Setup/wiki/Supported-Systems - seems silly to have the information in two places?

hhromic, 2019-03-20:
Agreed with @dankcushions , and also would do the same regarding the BIOS column. The job of that general table should be only to point to the relevant individual wiki pages where all the details are explained.

HerbFargus, 2019-03-20:
The idea was to have an overall view of the main requirements of each system but it's just one more think to keep parity with which I haven't had the free time to keep up with.  I'm fine with I being simplified, one less thing to maintain.

HNJAMeindersma, 2019-03-20:
I must say that I, personally, like the overview table on the Supported Systems page. However I must admit that I have never used it as a quick reference (I always go to the emu page directly).

If a choice has to be made that reducing the amount of redundancy would be the best fix for these kinds of errors, then I would agree with removing the data from the Supported Systems page.

However, then there would still be an error on the Commodore 64 page.
> **Commodore 64 / VIC 20**
> https://github.com/RetroPie/RetroPie-Setup/wiki/Commodore-64-VIC-20-PET - The emu table contains _.crt .d64 .g64 .t64 .tap .x64 **.zip** .prg_, however the ROM chapter contains _.crt .d64 .g64 .t64 .tap .x64 .prg_

joolswills, 2019-03-25:
The wiki is editable by anyone so feel free to correct anything you spot that is incorrect. Please use the forum for further discussion of wiki changes.

closed 2019-03-25 by joolswills