RetroPie/RetroPie-Setup #2665: Cannot update script since the lr-reicast commit

tacgnol3, 2019-03-27:
I get this error when I update :

                                               lqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqk
                                               x Update failed:                                           x
                                               x                                                          x
                                               x error: Your local changes to the following files would   x
                                               x be overwritten by merge:                                 x
                                               x       scriptmodules/libretrocores/lr-reicast.sh          x
                                               x h                                                        x
                                               x Please commit your changes or stash them before you      x
                                               x merge.                                                   x
                                               x Aborting                                                 x
                                               x                                                          x
                                               x                                                          x
                                               x                                                          x
                                               x                                                          x
                                               x                                                          x
                                               x                                                          x
                                               x                                                          x
                                               tqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqu
                                               x                         <  OK  >                         x
                                               mqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqj

cmitu, 2019-03-27:
You modified the script locally, as the error said. You'll have to remove your modifications, then the update will work again.

tacgnol3, 2019-03-27:
Nope, i never changed/modified anything. I only used the update function in the retropie-setup script to update the whole box.

joolswills, 2019-03-27:
Then the only other possibility is sdcard corruption. This is not a RetroPie-Setup issue though - please use the forum first as detailed when opening an issue here. Thanks.

closed 2019-03-27 by joolswills