RetroPie/RetroPie-Setup #2669: Updating DraStic to 2.5.0.4

DanTheMan827, 2019-03-30:
There's a Linux version of DraStic 2.5.0.4 that's fully SDL2 and fixes compatibility for games that use the microphone.

Is there any particular reason that this isn't used?

https://github.com/Retro-Arena/binaries/blob/master/odroid-xu4/drastic.tar.gz?raw=true

quicksilver7837, 2019-04-01:
There have been previous permission issues with distribution of drastic. I'll reach out to exophase (the author of drastic) on the drastic forums and see if he's ok with it.

DanTheMan827, 2019-04-01:
there's also xperia64

quicksilver7837, 2019-04-02:
https://drastic-ds.com/viewtopic.php?f=5&t=5610

I made a post on their forums. Lets hope I get a response.

robertybob, 2019-04-08:
Exophase said yes so that's great news!

2play, 2019-05-15:
@quicksilver7837 there was a request also and the devs forum and approved the use of that binary 
See here
https://drastic-ds.com/viewtopic.php?t=5610

mrcmunir, 2019-07-09:
If someone wants to try it under a pure distribution of aarch64 64bits for some devboards that are based on debian or ubuntu it's possible with multilib 

There workaround drastic run on aarch64

sudo dpkg --add-architecture armhf
sudo apt-get update
sudo apt-get install libc6: armhf libstdc ++ 6:armhf

Install small dependencies 
sudo apt-get install libsdl2-2.0-0:armhf
sudo apt-get install libasound2-plugins:armhf
sudo apt-get install zlib1g: armhf
sudo apt-get install libgtk3-nocsd0:armhf

joolswills, 2019-10-20:
This version is now included.

closed 2019-10-20 by joolswills

theofficialgman, 2020-05-15:
> If someone wants to try it under a pure distribution of aarch64 64bits for some devboards that are based on debian or ubuntu it's possible with multilib
>
> There workaround drastic run on aarch64 [...]

I've been able to follow this to run drastic on the nintendo switch but joystick detection doesn't seem to work. Does drastic even support sdl2 joysticks?

jksherk, 2020-10-23:
> @mrcmunir
> 
> > If someone wants to try it under a pure distribution of aarch64 64bits for some devboards that are based on debian or ubuntu it's possible with multilib
> > There workaround drastic run on aarch64
> > sudo dpkg --add-architecture armhf
> > sudo apt-get update
> > sudo apt-get install libc6:armhf libstdc++6:armhf
> > Install small dependencies
> > sudo apt-get install libsdl2-2.0-0:armhf
> > sudo apt-get install libasound2-plugins:armhf
> > sudo apt-get install zlib1g:armhf
> > sudo apt-get install libgtk3-nocsd0:armhf
> 
> I've been able to follow this to run drastic on the nintendo switch but joystick detection doesn't seem to work. Does drastic even support sdl2 joysticks?
I follow the step and get an error on    sudo apt-get install libsdl2-2.0-0:armhf
can't install , can you help ,thanks

cobalt2727, 2021-03-25:
> @mrcmunir
>
>
> I've been able to follow this to run drastic on the nintendo switch but joystick detection doesn't seem to work. Does drastic even support sdl2 joysticks?

bump

DanTheMan827, 2021-03-25:
Drastic supports SDL2 joysticks, but I don't know if it loads gamecontrollerdb.txt at all

It's definitely possible to write a LD_PRELOAD library that loads the gamecontrollerdb.txt though, but that would depend on if drastic even makes use of the gamepad interface or if it uses the joystick interface

joolswills, 2021-03-26:
@cobalt2727 this issue is closed. Please use the forum for support.