RetroPie/RetroPie-Setup #2677: In the logs folder there are actually 21 logs instead of 20

hiulit, 2019-04-09:
I know, it's a very silly "issue".

https://github.com/RetroPie/RetroPie-Setup/blob/307db2d5d14ac855abee1d722a19403acb438507/scriptmodules/admin/setup.sh#L91

It should be `head -n -19` instead of `head -n -20` to actually keep 20 logs.

hhromic, 2019-04-10:
Umm, the current script does work for me:
```
$ mkdir test && cd test
$ for f in {01..20}; do touch $f.log; done  # create 20 dummy files
$ find -type f | sort | head -n -20 | xargs -d '\n' --no-run-if-empty echo  # just print what would be deleted
(..nothing is shown, therefore nothing is deleted, 20 files are kept..)
$ for f in {01..21}; do touch $f.log; done  # create 21 dummy files (just one more from before)
$ find -type f | sort | head -n -20 | xargs -d '\n' --no-run-if-empty echo  # now we would delete the first one and keep the other 20 (02 to 21)
./01.log
```

Can you share the list of files in your `$__logdir` that triggers the issue for you?

hiulit, 2019-04-10:
Like this?

```
pi@raspberry:~/RetroPie-Setup/logs $ ls
rps_2019-03-27_134325.log.gz  rps_2019-04-02_152452.log.gz
rps_2019-03-27_134525.log.gz  rps_2019-04-02_152523.log.gz
rps_2019-03-27_134755.log.gz  rps_2019-04-02_153554.log.gz
rps_2019-04-02_134639.log.gz  rps_2019-04-02_153614.log.gz
rps_2019-04-02_135149.log.gz  rps_2019-04-02_153932.log.gz
rps_2019-04-02_135818.log.gz  rps_2019-04-02_154118.log.gz
rps_2019-04-02_135840.log.gz  rps_2019-04-02_154310.log.gz
rps_2019-04-02_140002.log.gz  rps_2019-04-02_154416.log.gz
rps_2019-04-02_140006.log.gz  rps_2019-04-02_154908.log.gz
rps_2019-04-02_140042.log.gz  rps_2019-04-02_161929.log.gz
rps_2019-04-02_140211.log.gz
```

hiulit, 2019-04-10:
I've run your tests and they also work for me.

But when I found this "issue" I was trying to use `find -type f | sort | head -n -20 | xargs -d '\n' --no-run-if-empty rm` on a project I'm working on and I also had to change it to `19`. That's why I brought it up, because I saw that I has 21 logs.

joolswills, 2019-04-10:
Probably because a new log is made after .. it's a non issue anyway imho.

hiulit, 2019-04-10:
Yeah, as I said it's indeed a silly "issue". If you want to close it that's fine.

I just brought it up in case there was something wrong with the code. But apparently there isn't anything wrong.

closed 2019-04-19 by joolswills